robertklep/dsm7-usb-serial-drivers #1: Can you make some for geminilake?

April-ha, 2021-07-12:
thanks

kennetn, 2021-07-12:
I tried with the apollolake on DS720+ but this shows the error: 
> xxxxxx@xxxxx:/lib/modules$ sudo insmod ch341.ko
> insmod: ERROR: could not insert module ch341.ko: Unknown symbol in module

I guess this is because it is not compiled for geminilake

robertklep, 2021-07-12:
Try these: https://github.com/robertklep/dsm7-usb-serial-drivers/tree/main/modules/geminilake

April-ha, 2021-07-12:
Hello Kenneth
Could you please indicate if it works with DS720+ before I switch to DSM 7
thank you

kennetn, 2021-07-12:
Thanks a lot.
I just tried again this is what I did.
>_nas_@_user_:/lib/modules$ sudo cp "/volume1/.../ch341.ko" ch341.ko
>_nas_@_user_:/lib/modules$ ls -l ch*
>---------- 1 root root 357264 Jul 12 17:46 ch341.ko
>_nas_@_user_:/lib/modules$ sudo chmod 744 ch341.ko
>_nas_@_user_:/lib/modules$ ls -l ch*
>-rwxr--r-- 1 root root 357264 Jul 12 17:46 ch341.ko
>_nas_@_user_:/lib/modules$ sudo insmod ch341.ko
>insmod: ERROR: could not insert module ch341.ko: Unknown symbol in module

I do not need to run the usb-serial-drivers.sh to test it right ?

robertklep, 2021-07-12:
@kennetn you may need to load one or more of these modules (mentioned in the script):
```
sudo insmod /lib/modules/usbserial.ko
sudo insmod /lib/modules/ftdi_sio.ko
sudo insmod /lib/modules/cdc-acm.ko
```

kennetn, 2021-07-12:
Now it looks like it loaded and I have a /dev/ttyUSB0

robertklep, 2021-07-12:
@kennetn nice! Thanks for testing 👍🏻

closed 2021-07-12 by robertklep