robertklep/dsm7-usb-serial-drivers #6: csr8510 bluetooth driver

Anonym-tsk, 2021-08-01:
Please make btusb.ko and bluetooth.ko for csr8510 bt dongle support.
At least for apollolake.

robertklep, 2021-08-01:
That's much more involved than just building a few serial drivers and very possibly will require compiling the entire kernel. Besides that, Synology hasn't published the kernel sources for DSM 7 yet, it's just lucky that the serial drivers haven't changed much in between kernel versions so I was able to pick them from an older kernel source tree.

IamMattM, 2022-06-20:
> Please make btusb.ko and bluetooth.ko for csr8510 bt dongle support. At least for apollolake.

I purchased the Sabrent BLE dongle <Bluetooth 4.0+LowEnergyMode> also based on "CSR8510 A10".
Ubuntu recognises it as Dongle "HCI" mode. 

usb 3-9: new full-speed USB device number 6 using xhci_hcd
usb 3-9: New USB device found, idVendor=0a12, idProduct=0001, bcdDevice=88.91
usb 3-9: New USB device strings: Mfr=0, Product=2, SerialNumber=0
usb 3-9: Product: CSR8510 A10

|__ Port 9: Dev 6, If 0, Class=Wireless, Driver=btusb, 12M
        ID 0a12:0001 Cambridge Silicon Radio, Ltd Bluetooth Dongle (HCI mode)
    |__ Port 9: Dev 6, If 1, Class=Wireless, Driver=btusb, 12M
        ID 0a12:0001 Cambridge Silicon Radio, Ltd Bluetooth Dongle (HCI mode)

However, the same dongle is not even registered on geminilake (DS920+) , DSM7

I was hoping to make use of docker.homeassistant.<ble2mqtt> integration to monitor BLE devices to complement my z-wave and zigbee ones.

robertklep, 2022-06-21:
Again, BT is a different beast from relatively simple USB serial drivers.

Perhaps [esp32-ble2mqtt](https://github.com/shmuelzon/esp32-ble2mqtt) is an acceptable alternative? It would require some additional hardware, but ESP32's are cheap.

IamMattM, 2022-06-21:
> Again, BT is a different beast from relatively simple USB serial drivers.
> 
> Perhaps [esp32-ble2mqtt](https://github.com/shmuelzon/esp32-ble2mqtt) is an acceptable alternative? It would require some additional hardware, but ESP32's are cheap.

Thank you.
Yes, I happen to have ordered an ESP32-based D1 mini  on which I might run Tasmota/ESPhome.

IamMattM, 2023-03-01:
It has been a while since querying on the matter: I thought to give an update to say that I have been running the "esp32-ble-gateway" service with mqtt and esphome support and it have been a good alternative.

robertklep, 2023-03-02:
Thanks for the update, @IamMattM 😊 And good to hear that you managed to get it working.

nmajin, 2023-03-21:
It seems someone has been able to compile these, I am wondering if its an easy thing to do in this case? https://community.home-assistant.io/t/solved-usb-bluetooth-support-on-synology-dsm-7-1/549248

I am using ESPHome bluetooth proxy as an alternative, but would like to just see if I can get native BT working with my Synology and my docker containers.

IamMattM, 2023-03-23:
> It seems someone has been able to compile these, I am wondering if its an easy thing to do in this case? [...]

@nmajin: THans for the pointer.
As I happen to run on DSM920+ (gemini lake) those kernel modules just worked.

However, the BlueZ stack and hci utilities on DSM7 were missing to properly make use of the CRS8510 now detected correctly as "Bluetooth Dongle (HCI mode) - Cambridge Silicon Radio, Ltd"

To cover the above feature, I installed entware for DSM7 and the optional bluetooth utilities following this post:
https://github.com/Entware/Entware/wiki/Install-on-Synology-NAS

Now able to see the BlueZ stack at version 5.66 using command:
"bluetoothd -v"
and confirm device is present:
"ls -l /sys/class/bluetooth"

As for integrating the BT dongle into HomeAssistant running in docker mode, I will be following this video and "mapping" "/run/dbus" to the container to present the BT dongle to Homeassistant:
https://www.youtube.com/watch?v=ZXvcdI3mQuA&t=23s

Saying this, because of the limited range of Bluetooth my preferred "system" olution is still to use the few ESP32 mini dotted around the house and running the ESPHome ble_gateway service but you never know...might need native bluetooth for some newer integrations....

Once again than you for the pointer.