robertklep/dsm7-usb-serial-drivers #7: usb-serial-drivers.sh is unnecessary

tomkv, 2021-08-04:
Just a heads-up: the loader script is not necessary.

DSM7 comes with systemd as the service manager. One of the supplied services is [systemd-modules-load.service](https://www.freedesktop.org/software/systemd/man/systemd-modules-load.service.html), which can ensure loading the modules.

So all you need is a single [.conf file](https://www.freedesktop.org/software/systemd/man/modules-load.d.html) containing a list of wanted modules in `/etc/modules-load.d/` and they will be loaded at boot (or whenever systemd-modules-load service is started).

robertklep, 2021-08-04:
@tomkv thanks for the tip, I'll add it to the README. Do you have any idea what happens when any of the modules that should be loaded from a `.conf` file doesn't exist? I'm trying to figure out if I can just provide a single file that contains all drivers, even if the user hasn't installed them all.

tomkv, 2021-08-04:
@robertklep Looking at the [source](https://github.com/systemd/systemd/blob/5efbd0bf897a990ebe43d7dc69141d87c404ac9a/src/shared/module-util.c), you will get logged message, if the module cannot be loaded, is on the blacklist or is already loaded.

You could also provide file - separate .conf - for each driver, and only what user installs,would get loaded.

robertklep, 2021-08-25:
For some reason, the systemd solution is failing (see #8). I'm not inclined to fix it at the moment so I'll remove the systemd-related instructions from the README.