robertklep/dsm7-usb-serial-drivers #8: Job for systemd-modules-load.service failed

abergui, 2021-08-24:
Hello, I am trying to install the driver on my Syno DS918+ (Apollolake) and I have follow these steps ::
- copy the required files to your Synology and move them to `/lib/modules`
- copy the file `90-usb-serial.conf` to `/usr/lib/modules-load.d/`

But when I ran `systemctl restart systemd-modules-load` to load the modules I experienced the following error ::
- Job for systemd-modules-load.service failed. See "systemctl status systemd-modules-load.service" and "journalctl -xe" for details.

journalctl -xe says ::
```
Aug 24 18:35:13 Syno918 sudo[8170]:     NTY: TTY=pts/0 ; PWD=/usr/lib/modules-load.d ; USER=root ; COMMAND=/b
in/systemctl restart systemd-modules-load
Aug 24 18:35:13 Syno918 sudo[8170]: pam_unix(sudo:session): session opened for user root by NTY(uid=1028)
Aug 24 18:35:13 Syno918 systemd[1]: Starting Load Kernel Modules...
Aug 24 18:35:13 Syno918 systemd-modules-load[8173]: Failed to find module 'cp210x'
Aug 24 18:35:13 Syno918 systemd-modules-load[8173]: Failed to find module 'ch341'
Aug 24 18:35:13 Syno918 systemd[1]: systemd-modules-load.service: main process exited, code=exited, status=1/FA
ILURE
Aug 24 18:35:13 Syno918 systemd[1]: Failed to start Load Kernel Modules.
Aug 24 18:35:13 Syno918 systemd[1]: Unit systemd-modules-load.service entered failed state.
Aug 24 18:35:13 Syno918 systemd[1]: systemd-modules-load.service failed.
```

But the 2 modules are really in `/lib/modules` and I have change them to `root:root` and `744`

Is someone have an idea to solve this ?
Thank you for your great job

robertklep, 2021-08-24:
The modules on my Syno have 644 file modes, perhaps that's the issue.

Reloading the service works fine for me:
```
$ sudo systemctl status systemd-modules-load
● systemd-modules-load.service - Load Kernel Modules
   Loaded: loaded (/usr/lib/systemd/system/systemd-modules-load.service; static; vendor preset: disabled)
   Active: active (exited) since Tue 2021-07-13 14:28:37 CEST; 1 months 11 days ago
     Docs: man:systemd-modules-load.service(8)
           man:modules-load.d(5)
 Main PID: 5502 (code=exited, status=0/SUCCESS)
   Memory: 0B
   CGroup: /system.slice/systemd-modules-load.service
```

abergui, 2021-08-24:
I change the 2 modules to 644 root:root but unfortunately not better

```
$ sudo systemctl status systemd-modules-load -l
● systemd-modules-load.service - Load Kernel Modules
   Loaded: loaded (/usr/lib/systemd/system/systemd-modules-load.service; static; vendor preset: disabled)
   Active: failed (Result: exit-code) since Tue 2021-08-24 20:50:46 CEST; 1min 34s ago
     Docs: man:systemd-modules-load.service(8)
           man:modules-load.d(5)
  Process: 12215 ExecStart=/usr/lib/systemd/systemd-modules-load (code=exited, status=1/FAILURE)
 Main PID: 12215 (code=exited, status=1/FAILURE)

Aug 24 20:50:46 Syno918 systemd[1]: Starting Load Kernel Modules...
Aug 24 20:50:46 Syno918 systemd[1]: systemd-modules-load.service: main process exited, code=exited, status=1/FAILURE
Aug 24 20:50:46 Syno918 systemd[1]: Failed to start Load Kernel Modules.
Aug 24 20:50:46 Syno918 systemd[1]: Unit systemd-modules-load.service entered failed state.
Aug 24 20:50:46 Syno918 systemd[1]: systemd-modules-load.service failed.
```

robertklep, 2021-08-25:
Hmm yeah you're right, I didn't actually reload the service. Now that I do, I'm getting the same error:
```
Aug 25 08:02:43 syno systemd-modules-load[3997]: Failed to find module 'cp210x'
```

I'll remove the systemd-related instructions from the README, I'm not inclined to dive into the idiosyncrasies of systemd at the moment.

Thanks for your help!

closed 2021-08-25 by robertklep

GravityRZ, 2023-03-10:
maybe a dumb question but can anybody check if the directory needs to be

/etc/modules-load.d/  instead of /usr/lib/modules-load.d/

robertklep, 2023-03-10:
@GravityRZ `/etc/modules-load.d` doesn't exist on my NAS, but `/usr/lib/modules-load.d` does (and contains Synology-specific files). But see [this comment](https://github.com/robertklep/dsm7-usb-serial-drivers/issues/7#issuecomment-947586483).

GravityRZ, 2023-03-10:
ok .thanks