robertklep/dsm7-usb-serial-drivers #17: ti_usb_3410_5052 for Braswell

wkraaij, 2022-01-03:
Hi Robert,

can you compile this module for DSM 7 (it was available in the Jadahl drivers for DSM6.2 but not in DSM 7 for my architecture (Braswell)

thanks!

Wessel

robertklep, 2022-01-03:
@wkraaij hope this works: https://github.com/robertklep/dsm7-usb-serial-drivers/tree/main/modules/braswell

wkraaij, 2022-01-03:
Thank a lot!, I can at least load the driver in the kernel. But the ttyUSB device is not created yet. I need to tinker a bit more. I know I need to load a firmware file as well. Will report...

wkraaij, 2022-01-03:
I added a rule in /lib/udev/rules.d to properly initialize the module and put the firmware file in /lib/firmware. Now my evohome sensors are visible again in Domoticz!

robertklep, 2022-01-03:
Cool, and thanks for your donation! 😊