robertklep/dsm7-usb-serial-drivers #20: MCE USB IR receiver support?

cncb-gh, 2022-01-31:
Sorry for the off-topic question, but I was able to get my Z-Wave USB dongle working perfectly with your instructions.  Do you know if there is something similar that can be done for a generic MCE USB infrared receiver?  It is recognized without any special drivers in most other Linux installs.  Thanks.

robertklep, 2022-01-31:
The nice thing about these USB serial drivers is that most of their support is already provided by DSM 7 (`usbserial.ko`), so all I have to do is to compile the individual USB chipset drivers.

MCE devices depend on all the "RC core" modules and `mceusb.ko`, which is much more involved to get working (I've tried something similar with BT/BLE drivers but gave up). At some point I'll try and see if I can build an entire Linux kernel for DSM and cherrypick the drivers from there, but that's something for the future.

cncb-gh, 2022-01-31:
OK, thank you.

closed 2022-01-31 by cncb-gh