robertklep/dsm7-usb-serial-drivers #21: insmod: Invalid module format

caco3, 2022-02-26:
Just a note:
I have a DSM7 setup on an appololake system (Xpenology, DS918+)
I want to use the driver for ch341, but I get:
```
insmod /lib/modules/ch341.ko
insmod: ERROR: could not insert module /lib/modules/ch341.ko: Invalid module format
```

Then I went to http://www.jadahl.com/iperf-arp-scan/DSM_7.0/apollolake/ and tried the version from there which loaded and works without any error.
So there seems to be some odity on the driver in the repo here.

robertklep, 2022-02-27:
Strange, I have the exact same hardware and it's working for me (using the file from Github):
```
$ uname -a
Linux syno 4.4.180+ #42218 SMP Mon Oct 18 19:17:56 CST 2021 x86_64 GNU/Linux synology_apollolake_918+
$ sudo insmod ch341.ko
Password:
$ sudo lsmod
Module                  Size  Used by
ch341                   5369  0
```

swede89, 2023-01-14:
Hi. I face the same issue like @caco3. 

`# insmod ch341.ko
insmod: ERROR: could not insert module ch341.ko: Invalid module format`

My system is a 218+
`# uname -a
Linux DiskStation 4.4.180+ #42962 SMP Tue Oct 18 15:05:01 CST 2022 x86_64 GNU/Linux synology_apollolake_218+`

`# dmesg | tail -5
[45559.083505] ch341: Unknown symbol usb_serial_handle_dcd_change (err 0)
[45559.090828] ch341: Unknown symbol usb_serial_generic_resume (err 0)
[45559.097853] ch341: Unknown symbol usb_serial_deregister_drivers (err 0)
[45559.105280] ch341: Unknown symbol usb_serial_generic_tiocmiwait (err 0)
[45559.112694] ch341: Unknown symbol usb_serial_register_drivers (err 0)`

I was hoping to gain support for my USB dongle with CH340 chipset, though this file indicates CH341 support. @robertklep, can you please provide a module for my chipset?

Thanks

Simon

robertklep, 2023-01-15:
@swede89 did you load the `usbserial` module prior to loading the CH341 module?
```
insmod /lib/modules/usbserial.ko
```

leafy911, 2025-11-06:
Hi Robert

Thanks for your quick response. I looked through my history and these were the commands that I ran:

**_sudo wget https://github.com/robertklep/dsm7-usb-serial-drivers/blob/main/modules/denverton/dsm-7.2/ch341.ko
sudo wget https://github.com/robertklep/dsm7-usb-serial-drivers/blob/main/modules/denverton/dsm-7.2/cp210x.ko
sudo wget https://github.com/robertklep/dsm7-usb-serial-drivers/blob/main/modules/denverton/dsm-7.2/cp210x.ko
sudo wget https://github.com/robertklep/dsm7-usb-serial-drivers/blob/main/modules/denverton/dsm-7.2/pl2303.ko
sudo wget https://github.com/robertklep/dsm7-usb-serial-drivers/blob/main/modules/denverton/dsm-7.2/ti_usb_3410_5052.ko_**

Thanks in advance.

robertklep, 2025-11-06:
@leafy911 

The 7.2 modules all declare 4.4.302 as version magic, which is the correct one for your platform. The 7.0 and 7.1 modules declare 4.4.180 as version magic, which is the version you're getting an error about. 

Can you post the result of this command?

```
grep -o --binary-files=text 'vermagic=.*mod_' /lib/modules/ch341.ko
```