robertklep/dsm7-usb-serial-drivers #23: cp210x.ko Unknown symbol in module: DS720+ geminilake

khvMX, 2022-03-25:
Current DSM Version: DSM 7.0.1-42218 Update 3

sh-4.4$ uname -a
`Linux mushkoNAS 4.4.180+ #42218 SMP Mon Oct 18 19:16:55 CST 2021 x86_64 GNU/Linux synology_geminilake_720+`

sh-4.4$ sudo insmod /lib/modules/cp210x.ko 
`insmod: ERROR: could not insert module /lib/modules/cp210x.ko: Unknown symbol in module`

dmesg output:
[ 2610.398136] cp210x: module verification failed: signature and/or required key missing - tainting kernel
[ 2610.408753] cp210x: Unknown symbol usb_serial_generic_open (err 0)
[ 2610.415673] cp210x: Unknown symbol usb_serial_generic_close (err 0)
[ 2610.422700] cp210x: Unknown symbol usb_serial_deregister_drivers (err 0)
[ 2610.430196] cp210x: Unknown symbol usb_serial_register_drivers (err 0)

khvMX, 2022-03-25:
@robertklep hi, any tips? what else could I try?

robertklep, 2022-03-25:
@khvMX the `cp210x` driver depends on the `usbserial` driver to be loaded first:
```
insmod /lib/modules/usbserial.ko
```

khvMX, 2022-03-25:
oh, somehow I though I did that, but after reboot it works now, thanks!

closed 2022-03-25 by khvMX