robertklep/dsm7-usb-serial-drivers #28: i have deleted serialusb.ko

vocaris9, 2022-07-17:
Sorry,
but I accidentally deleted the file serialisb.ko mit rm.
I habe a DS220+
Where can i find/download this file?

robertklep, 2022-07-17:
It's part of DSM itself. If you have DSM updates pending, that might be a way to get it back, otherwise find someone with the same architecture ([Geminilake](https://kb.synology.com/en-uk/DSM/tutorial/What_kind_of_CPU_does_my_NAS_have)) to send you the file from their Synology (I have Apollolake so can't help there, I'm afraid).

vocaris9, 2022-07-17:
OK. Thank you.

baylanger, 2022-07-17:
@vocaris9 

```
# uname -a
Linux san 4.4.180+ #42661 SMP Mon Jun 27 15:07:20 CST 2022 x86_64 GNU/Linux synology_geminilake_720+
# ls -l /usr/lib/modules/usbserial.ko
-rw-r--r-- 1 root root 46875 Apr  1 05:53 /usr/lib/modules/usbserial.ko
# md5sum /usr/lib/modules/usbserial.ko
150581651ccc3b83157cb10665761584  /usr/lib/modules/usbserial.ko

GitHub didn't accept a .ko file ... so I was lazy and renamed it as a .pdf file.  Only once I post this reply that I'll know if the file is downloaded...  MAKE SURE to check md5sum matches before moving it into the "kernel modules" directory.  You should have make this file owned by root:root w/ 644 perms (like other files already in there).

```
[image]

baylanger, 2022-07-17:
It works.

From my Mac: 
```
% curl https://user-images.githubusercontent.com/5240348/179404307-3f2867d4-2b35-47f6-9aea-440f707ad117.jpeg -o usbserial.ko
% md5 usbserial.ko
MD5 (usbserial.ko) = 150581651ccc3b83157cb10665761584
```

vocaris9, 2022-07-17:
@baylanger Thank you. It runs...

robertklep, 2022-07-17:
@baylanger nice workaround! And thanks for the help 😁

closed 2022-07-17 by robertklep

dominicsherry1, 2024-04-29:
I’ve made the same mistake but on DSM7.2.1-69057 update 5. @baylanger ’s file is for 7.1.1 I believe. Does anyone have a geminilake device with the latest update and able to share serialusb.ko with me please?

dominicsherry1, 2024-04-29:
@Koky05 Can you rename it to png and attach it to a comment on this thread?

Koky05, 2024-04-29:
[image]

dominicsherry1, 2024-04-29:
It works. Thanks a lot!