robertklep/dsm7-usb-serial-drivers #29: How to troubleshoot?

thyeestes, 2022-07-28:
Hi,

I followed the instructions, but I don't see my USB serial device in my DS216Play NAS. I there some guidance available on how to troubleshoot in DSM7?

Cheers,
Martijn

robertklep, 2022-07-28:
What's the output of these two commands?
* `lsmod | grep '^usbcore'`
* `lsusb -cui`

thyeestes, 2022-07-28:
root@DS216PLAY:~# lsmod |grep '^usbcore'
usbcore               141711  11 ch341,usblp,usb_storage,cp210x,usbserial,ohci_hcd,ehci_hcd,ehci_pci,usbhid,xhci_hcd,ehci_platform

root@DS216PLAY:~# lsusb -cui
|__usb2          1d6b:0001:0310 09  1.10   12MBit/s 0mA 1IF  (Linux 3.10.108 ohci_hcd Generic Platform OHCI Controller ohci-platform.0) hub
  |__2-1         0403:6001:0600 00  2.00   12MBit/s 90mA 1IF  (FTDI FT232R USB UART AB5H6R5G)
  2-1:1.0         (IF) ff:ff:ff 2EPs (Vendor Specific)

robertklep, 2022-07-28:
I'm missing the `ftdi_sio` driver in the first list, which is probably the one that's required for your device. You don't need my drivers for it, they are included in DSM (just not loaded).

Try `insmod /lib/modules/ftdi_sio.ko`

thyeestes, 2022-07-28:
That solved the issue.

Thanks mate!

Kees-Schouten, 2023-01-06:
Thanks! That solved the issue for me as well.

closed 2023-01-07 by robertklep