robertklep/dsm7-usb-serial-drivers #32: PL2303 for cedarview (RS814+)

miticolars, 2022-08-22:
Hi, can you provide the dms7 driver for cedarview platform? I have to connect a pl2303 usb serial converter used by homebridge (docker). Thanks in advance.

robertklep, 2022-08-22:
I'll try and see if I'm able to compile it, I'm having some issues with my dev VM not wanting to start up anymore 😬

miticolars, 2022-08-22:
ok, thank you!

robertklep, 2022-08-22:
One question: what's the current Linux kernel version of your NAS for DSM 7?

miticolars, 2022-08-22:
I'm sorry, but I haven't updated the DSM yet

robertklep, 2022-08-22:
I understand 😊 But I need the kernel version to be able to compile the correct version of the driver. I'll see if I can find out some other way.

robertklep, 2022-08-22:
FWIW, it's possible that one of the other drivers (most likely `apollolake` or `geminilake`) will work on your hardware too, but I guess you first need to update your DSM to be able to try that (and there are no guarantees it'll work).

miticolars, 2022-08-22:
Ok, when I'll update my system I will try the other drivers and I will back to you.

robertklep, 2022-08-22:
I still can't guarantee I'll be able to compile it within a reasonable time period, so please only update if you're okay with not being able to use your device for a while.

miticolars, 2022-08-22:
ok, thanks!

baylanger, 2022-08-22:
@miticolars

Unless Synology made new version available in past few days that hasn't been pushed worldwide yet, this is with latest DSM7.

{noformat}
Linux san 4.4.180+ #42661 SMP Mon Jun 27 15:07:20 CST 2022 x86_64 GNU/Linux synology_geminilake_720+
{noformat}

baylanger, 2022-08-22:
Deleted my previous comment, I got side tracked :)

robertklep, 2022-08-22:
@baylanger no worries 😅

robertklep, 2022-08-23:
Looks like it's 3.10.x: https://archive.synology.com/download/ToolChain/Synology%20NAS%20GPL%20Source/7.0-41890

In which case the Apollolake and Geminilake modules won't work, those are for kernel version 4.4.x. But other platforms using 3.10.x are `avoton`, `braswell` and `monaco`.

miticolars, 2022-12-04:
Hi @robertklep, finally I installed and tested your pl2303 driver on DSM 7.1 on RS814+ (cedarview) Synology NAS and it works! Thank you!