robertklep/dsm7-usb-serial-drivers #34: Drivers for R1000 architecture

ilco13, 2022-09-14:
Hello,

Can you provide the drivers for R1000 (Ryzen R1600 on RS422 & DS1522) architecture ?

Maybe it's the same as V1000 but....?

robertklep, 2022-09-15:
It was reported in #31 that drivers for V1000 also work on R1000, although that's the other way around from what you're asking. It's pretty safe to try them, if they aren't the correct platform the kernel won't load them.

bartcerneels, 2022-12-11:
I've just tried [the v1000 cp210x module](https://github.com/robertklep/dsm7-usb-serial-drivers/blob/main/modules/v1000/dsm-7.1/cp210x.ko) on a DS923+ but it will not load:

```
bart@RackStation:~$ wget https://github.com/robertklep/dsm7-usb-serial-drivers/raw/main/modules/v1000/dsm-7.1/cp210x.ko

bart@RackStation:~$ sudo insmod cp210x.ko
Password:
insmod: ERROR: could not insert module cp210x.ko: Unknown symbol in module
```

robertklep, 2022-12-11:
Try these: https://github.com/robertklep/dsm7-usb-serial-drivers/tree/main/modules/r1000/dsm-7.1

bartcerneels, 2022-12-11:
Turns out they do work, just needed the dependencies loaded first. 

And in the end, turns out I need an FTDI driver and it's already built into DSM 7.1.

```
bart@RackStation:/usr/lib/modules$ lsmod | grep serial
usbserial              21462  4 ch341,cp210x,pl2303,ftdi_sio
usbcore               200268  14 ch341,etxhci_hcd,usblp,uhci_hcd,usb_storage,cp210x,usbserial,ehci_hcd,ehci_pci,pl2303,usbhid,ftdi_sio,xhci_hcd,xhci_pci
```

bartcerneels, 2022-12-11:
> Try these: https://github.com/robertklep/dsm7-usb-serial-drivers/tree/main/modules/r1000/dsm-7.1

I can confirm these load.
```
sh-4.4# insmod ch341.ko
sh-4.4# insmod cp210x.ko
sh-4.4# insmod pl2303.ko
sh-4.4# lsmod | grep usb
usbserial              21462  4 ch341,cp210x,pl2303,ftdi_sio
```

robertklep, 2022-12-12:
😂

I did compile them specifically for R1000 but I guess that the DSM toolkits for R1000 and V1000 are the same (which I didn't check for).

Anyway, good to read you got it working 👍🏻

closed 2022-12-12 by robertklep