robertklep/dsm7-usb-serial-drivers #35: insmod: Invalid module format on DS218+/Apollolake

nmehnert73, 2022-10-07:
Using insmod with the cp210.ko leads to the error: Invalid module format.

I've read that in some comparable aces the version from Jadahl did work - but unfortunately it is not longer availabale

Anyone an idea?

robertklep, 2022-10-07:
I have an ApolloLake platform (DS918+) myself, although I'm not on the latest version of DSM (moving away from Synology) so it might be caused by a kernel version mismatch.

You can still find the Jadahl drivers here: https://web.archive.org/web/20220126180837/http://www.jadahl.com/iperf-arp-scan/DSM_7.0/apollolake/

vytautask, 2022-10-21:
@nmehnert73 you probably need to recompile it for DSM 7.1. At least I've had to for geminilake platform to work with latest DSM...

robertklep, 2023-08-30:
> insmod: ERROR: could not insert module /lib/modules/ch341.ko: Invalid module format

Follow the downloading instructions _carefully_.

closed 2023-08-30 by robertklep