robertklep/dsm7-usb-serial-drivers #37: PL2303 Apollolake

erich74, 2022-10-20:
Hi,
Thanks a lot for putting those resources available!
Could you please add PL2303 available for Apollolake (DS218+ for me) please?
I have a Zigate Zigbee (v1) using this serial interface, I could make it work using the following link https://web.archive.org/web/20220126180837/http://www.jadahl.com/iperf-arp-scan/DSM_7.0/apollolake/

but it would be nice if you could add it so that we can only keep your GH as a single source and reference!
Thanks a lot

robertklep, 2022-10-21:
Done! I'll do the same for the other platforms, it's a shame that the Jadahl site is offline 😕

closed 2022-10-21 by robertklep

erich74, 2022-10-21:
Thank you very much Robert. Yes this is a shame but we have your GH now as the reference ;)