robertklep/dsm7-usb-serial-drivers #39: Error with DSM 7.1

viserion, 2022-10-21:
if i want to use the cp2102 driver with DSM 7.1.1-42962 Update 1  on  the Avoton Platform (DS 1817+) i get the following error:

stty: /dev/ttyUSB0: Inappropriate ioctl for device

driver is loaded:
lsmod:
usbcore               179491  14 ch341,etxhci_hcd,usblp,uhci_hcd,usb_storage,cp210x,usbserial,ehci_hcd,ehci_pci,pl2303,usbhid,ftdi_sio,cdc_acm,xhci_hcd

lsusb:
|__usb2          1d6b:0002:0310 09  2.00  480MBit/s 0mA 1IF  (Linux 3.10.108 etxhci_hcd-170202 Etron xHCI Host Controller 0000:04:00.0) hub
  |__2-2         10c4:ea60:0100 00  1.10   12MBit/s 100mA 1IF  (Silicon Labs CP2102 USB to UART Bridge Controller 0040)

before the update to 7.1 it was working fine. 
could you please provide a new driver for DMS 7.1?

Best regards

robertklep, 2022-10-21:
Can you check the kernel version for 7.1? Should be in `uname -a`.

viserion, 2022-10-21:
Linux DiskStation 3.10.108 #42962 SMP Wed Sep 21 10:54:35 CST 2022 x86_64 GNU/Linux synology_avoton_1817+

robertklep, 2022-10-21:
Try and see if these are working: https://github.com/robertklep/dsm7-usb-serial-drivers/tree/main/modules/avoton/dsm-7.1

viserion, 2022-10-21:
the error is gone, thanks. 
anyway it`s not working, I cant get data from the usb ir-head.

robertklep, 2022-10-21:
Thanks for testing. I assume it worked before (on 7.0)?

viserion, 2022-10-21:
it`s working now. my skript to receive the data has some problems, but the communication works!
Thanks for your help!

yes, it worked before

robertklep, 2022-10-21:
Cool! Thanks for the update 👍🏻

closed 2022-10-21 by robertklep