robertklep/dsm7-usb-serial-drivers #40: Building for DSM 7.1

Daanoz, 2022-10-21:
Unexpectedly broke my setup at home by updating DSM to 7.1, my zigbee dongle no longer works... In hindsight not a great move, since half the home now has no light anymore 🤦‍♂️ . After some googling I found myself here, where already some great work has been for DSM 7.0.

Anyway, the old modules are no longer loading:
```
insmod: ERROR: could not insert module /lib/modules/ch341.ko: Invalid module format
```

Since the instructions for building seem quite doable, tried to build the new modules for 7.1 myself, but hit a bit of snag there. Maybe someone knows what I'm doing wrong. My Synology is a DS216Play, so the `braswell` platform. If I understand correctly, after setting up the environment, to build the modules the command would be:
```
sudo toolkit/pkgscripts-ng/PkgCreate.py -X 1 -P 1 -v 7.1 --min-sdk 7.1 -p braswell 4.4.x
```

But it fails with:
```
scripts/Makefile.build:314: recipe for target '//source/4.4.x/pl2303.o' failed
make[2]: *** [//source/4.4.x/pl2303.o] Error 1
Makefile:1229: recipe for target '_module_//source/4.4.x' failed
make[1]: *** [_module_//source/4.4.x] Error 2
make[1]: Leaving directory '/usr/local/x86_64-pc-linux-gnu/x86_64-pc-linux-gnu/sys-root/usr/lib/modules/DSM-7.1/build'
Makefile:8: recipe for target 'ch341.ko' failed
make: *** [ch341.ko] Error 2
[Error] Build project fail!
Time cost: 00:00:02 [Build-->4.4.x]
[INFO] Build 4.4.x finished!
I got the following error:
======  Find pattern [Error] ======
15: make[2]: *** [//source/4.4.x/ch341.o] Error 1
17: make[1]: *** [_module_//source/4.4.x] Error 2
20: make: *** [cp210x.ko] Error 2
40: make[2]: *** [//source/4.4.x/pl2303.o] Error 1
42: make[1]: *** [_module_//source/4.4.x] Error 2
45: make: *** [ch341.ko] Error 2
46: [Error] Build project fail!
======  Find pattern [error:] ======
26: //source/4.4.x/pl2303.c:556:8: error: implicit declaration of function 'C_CMSPAR'; did you mean 'CMSPAR'? [-Werror=implicit-function-declaration]
31: //source/4.4.x/pl2303.c:755:18: error: 'struct usb_serial_port' has no member named 'minor'
34: //source/4.4.x/pl2303.c:756:20: error: 'struct usb_serial_port' has no member named 'port_number'; did you mean 'number'?
```

Something that I'm not doing correct here? I'm running this by the way in a ubuntu 18 VM on my desktop, instead of on the Synology since it's not super powerful.

robertklep, 2022-10-22:
Hoi @Daanoz!

According to [this page](https://kb.synology.com/en-uk/DSM/tutorial/What_kind_of_CPU_does_my_NAS_have), the DS216Play actually uses platform `monaco`, which could explain the format error (Braswell is Intel-based, Monaco uses an STM CPU).

Try and see if the [Monaco drivers](https://github.com/robertklep/dsm7-usb-serial-drivers/tree/main/modules/monaco/dsm-7.0) work for you. If not, I'll build them for 7.1 (the building process is a bit of a can of worms, it always errors out for me too but still builds the .ko files).

Daanoz, 2022-10-22:
Ah, that's on me, I mistyped there... I actually have the DS416Play.

Would be great if you could do a build, as I don't really know how to get it solved.

robertklep, 2022-10-22:
Check and see if these work for you: https://github.com/robertklep/dsm7-usb-serial-drivers/tree/main/modules/braswell/dsm-7.1

Daanoz, 2022-10-22:
Awesome! Works like a charm!

Me and my family thank you 👍

closed 2022-10-22 by Daanoz

robertklep, 2022-10-22:
Thanks for the donation!