robertklep/dsm7-usb-serial-drivers #41: insmod: cp210x.ko: Invalid module format (Apollolake - DS918+ - DSM 7.1.1-42962 Update 2)

serkanuto, 2022-10-25:
Using insmod with the cp210.ko leads an error.
I use the version for DSM 7.1.

`insmod: ERROR: could not insert module cp210x.ko: Invalid module `format`

robertklep, 2022-10-26:
Can you run this command and post its output?
```
file /lib/modules/cp210.ko
```

maxbec, 2022-10-26:
/lib/modules/cp210x.ko: HTML document, UTF-8 Unicode text, with very long lines

robertklep, 2022-10-26:
It shouldn't be HTML 😂 

You need to download the file as binary, [try this link](https://github.com/robertklep/dsm7-usb-serial-drivers/raw/main/modules/apollolake/dsm-7.1/cp210x.ko)

maxbec, 2022-10-26:
how can i tell wget to download as binary?

yverry, 2022-11-01:
you can close this issue. Problem is HTML was downloaded instead of the blog file