robertklep/dsm7-usb-serial-drivers #44: USB audio drivers?

TemplarB, 2022-10-30:
Hi!
I wanted to ask, what are the specific USB drivers for audio and how to install them? Formerly I used USB output from Synology's Audio Station to Topping E30 DAC. I installed drivers from here but with hasn't turned audio on. From what I understand the DAC uses XMOS XU208 chip (but I may err)

robertklep, 2022-10-30:
The modules in this repository are for USB serial devices only, I'm afraid.

TemplarB, 2022-10-30:
Thanks for the speedy reply! Where should I search for USB audio drivers? (Google hasn't helped much, maybe I don't ask it the right way)

robertklep, 2022-10-30:
I also can't find USB audio drivers for DSM, and I think it's safe to say that they simply don't exist.

You could possibly get your DAC to work using a proper Linux inside a virtual machine on your NAS.

TemplarB, 2022-10-30:
Thanks for the advice! I plan to, but I hoped for a simpler option first