robertklep/dsm7-usb-serial-drivers #48: Denverton 7.1

hsharrison, 2022-11-15:
Just setting up my Zigbee and ran into this issue. Can you build for Denverton 7.1? Thanks.

robertklep, 2022-11-16:
What kernel version is used on that platform? `uname -a`

hsharrison, 2022-11-16:
```
Linux diskstation 4.4.180+ #42962 SMP Tue Oct 18 15:02:07 CST 2022 x86_64 GNU/Linux synology_denverton_1618+
```

robertklep, 2022-11-16:
Try these: https://github.com/robertklep/dsm7-usb-serial-drivers/tree/main/modules/denverton/dsm-7.1

hsharrison, 2022-11-16:
Thanks for the fast response!

I'm not sure it's working though. At least I can't get my Conbee II to show up when I try to add ZHA in HomeAssistant. But I've never done that before, so there's a chance I'm doing something wrong. I understand from seeing screenshots that it should autodiscover but it's making me type in the device path, which I'm having trouble finding.

More to the point. is there a way to check that it's working? From `dmesg` or something? I'm pretty sure I did everything right, I live in the terminal, but don't know the first thing about kernel or devices.

hsharrison, 2022-11-16:
Saw your response in another issue about troubleshooting.

```
admin@diskstation:~$ lsusb
|__usb1          1d6b:0002:0404 09  2.00  480MBit/s 0mA 1IF  (Linux 4.4.180+ xhci-hcd xHCI Host Controller 0000:00:15.0) hub
  |__1-1         f400:f400:0100 00  2.00  480MBit/s 200mA 1IF  (Synology DiskStation 7F0084D45C6DAE01)
  |__1-2         0bda:5411:0104 09  2.10  480MBit/s 0mA 1IF  (Generic 4-Port USB 2.0 Hub syno.ext.hub) hub
    |__1-2.2     0658:0200:0000 02  2.00   12MBit/s 100mA 2IFs ( ffffffd1ffffffb2ffffffdbffffffad)
    |__1-2.3     1cf1:0030:0100 02  2.01   12MBit/s 100mA 2IFs (dresden elektronik ingenieurtechnik GmbH ConBee II DE2661119)
|__usb2          1d6b:0003:0404 09  3.00 5000MBit/s 0mA 1IF  (Linux 4.4.180+ xhci-hcd xHCI Host Controller 0000:00:15.0) hub
  |__2-2         0bda:0411:0104 09  3.00 5000MBit/s 0mA 1IF  (Generic 4-Port USB 3.0 Hub syno.ext.hub) hub
admin@diskstation:~$ lsmod | grep '^usbcore'
usbcore               203150  15 ch341,etxhci_hcd,usblp,uhci_hcd,usb_storage,cp210x,usbserial,ehci_hcd,ehci_pci,pl2303,usbhid,ftdi_sio,cdc_acm,xhci_hcd,xhci_pci
admin@diskstation:~$ lsusb -cui
|__usb1          1d6b:0002:0404 09  2.00  480MBit/s 0mA 1IF  (Linux 4.4.180+ xhci-hcd xHCI Host Controller 0000:00:15.0) hub
  |__1-1         f400:f400:0100 00  2.00  480MBit/s 200mA 1IF  (Synology DiskStation 7F0084D45C6DAE01)
  1-1:1.0         (IF) 08:06:50 2EPs () usb-storage host8 (synoboot)
  |__1-2         0bda:5411:0104 09  2.10  480MBit/s 0mA 1IF  (Generic 4-Port USB 2.0 Hub syno.ext.hub) hub
    |__1-2.2     0658:0200:0000 02  2.00   12MBit/s 100mA 2IFs ( ffffffd1ffffffb2ffffffdbffffffad)
   1-2.2:1.0      (IF) 02:02:01 1EP  () cdc_acm tty/ttyACM0
   1-2.2:1.1      (IF) 0a:00:00 2EPs () cdc_acm
    |__1-2.3     1cf1:0030:0100 02  2.01   12MBit/s 100mA 2IFs (dresden elektronik ingenieurtechnik GmbH ConBee II DE2661119)
   1-2.3:1.0      (IF) 02:02:01 1EP  () cdc_acm tty/ttyACM1
   1-2.3:1.1      (IF) 0a:00:00 2EPs () cdc_acm
|__usb2          1d6b:0003:0404 09  3.00 5000MBit/s 0mA 1IF  (Linux 4.4.180+ xhci-hcd xHCI Host Controller 0000:00:15.0) hub
```

Giving ZHA `/dev/ttyACM1` didn't work so I guess I'm missing a module.

robertklep, 2022-11-16:
@hsharrison it looks okay to me.

Are you running HA in Docker? If so, are you passing the correct device entry to it? Also, check the HA logs to see if there's any information on why it's not working.

hsharrison, 2022-11-16:
Yeah, I'm using docker. I added the devices to the JSON but didn't work until I enabled privileged mode. Is that typically necessary or it means I didn't map the devices correctly?

hsharrison, 2022-11-17:
Ah, I didn't realize you could map devices as volumes. Got it working! Thanks for your help.

Closed by 3d800611