robertklep/dsm7-usb-serial-drivers #49: cedarview driver for Tead Sonoff Zigbee 3.0 USB Dongle Plus V2 model "ZBDongle-E"

fruitvlieg1, 2022-11-29:
Hi Robert,
I'm looking for a driver of DSM DS1513+ to get the USB Dongle working.

Cedarview driver for Tead Sonoff Zigbee 3.0 USB Dongle Plus V2 model "ZBDongle-E"
dsm7.1-usb-serial-drivers 
firmware for Silicon Labs EFR32MG21

robertklep, 2022-11-30:
Because `/dev/ttyACM0` exists, there's already a driver installed for it 😊

fruitvlieg1, 2022-11-30:
Hi Robert thx for the reply!

Ok, there is a driver already installed. what driver is that the new one?
but /dev/ttyUSB0 is not there?
When continue with installation i received the error
chmod 777 /dev/ttyUSB0 doesn't exits 

/sbin/modprobe usbserial
/sbin/modprobe ftdi_sio
/sbin/modprobe cdc-acm
chmod 777 /dev/ttyUSB0
chmod 777 /dev/ttyACM0

And i get stuck on the Zigbee Integration:
What should i enter?
 
Radio Type:
EZSP = Silicon Labs EmberZNet protocol: Elelabs, HUSBZB-1, Telegesis

Serial device path:  <required>

Port speed: 57600 (default)

Data Flow control (software default)

robertklep, 2022-11-30:
You probably don't need `/dev/ttyUSB0` if you already have `/dev/ttyACM0`.

If it doesn't work, can you post the output of `lsusb -vt`?

fruitvlieg1, 2022-11-30:
option:  -vt  not supported

root@DiskStation:~# lsusb -vt
(<class 'getopt.GetoptError'>, GetoptError('option -v not recognized', 'v'), <traceback object at 0x7fbd74bcb6c0>)
Usage: lsusb.py [options]
Options:
 -h display this help
 -i display interface information
 -I display interface information, even for hubs
 -u suppress empty hubs
 -U suppress all hubs
 -c use colors
 -e display endpoint info
 -w display warning if usb.ids is not sorted correctly
 -f FILE override filename for /usr/share/usb.ids
Use lsusb.py -ciu to get a nice overview of your USB devices.
root@DiskStation:~#

fruitvlieg1, 2022-11-30:
root@DiskStation:~# lsusb -vt
(<class 'getopt.GetoptError'>, GetoptError('option -v not recognized', 'v'), <traceback object at 0x7fbd74bcb6c0>)
Usage: lsusb.py [options]
Options:
 -h display this help
 -i display interface information
 -I display interface information, even for hubs
 -u suppress empty hubs
 -U suppress all hubs
 -c use colors
 -e display endpoint info
 -w display warning if usb.ids is not sorted correctly
 -f FILE override filename for /usr/share/usb.ids
Use lsusb.py -ciu to get a nice overview of your USB devices.
root@DiskStation:~#

robertklep, 2022-11-30:
`lsusb -iI` then

fruitvlieg1, 2022-11-30:
root@DiskStation:~# lsusb -iI
|__usb1          1d6b:0002:0310 09  2.00  480MBit/s 0mA 1IF  (ehci_hcd 0000:00:1a.7)
 1-0:1.0          (IF) 09:00:00 1EP  () hub
|__usb2          1d6b:0002:0310 09  2.00  480MBit/s 0mA 1IF  (ehci_hcd 0000:00:1d.7)
 2-0:1.0          (IF) 09:00:00 1EP  () hub
  |__2-1         f400:f400:0100 00  2.00  480MBit/s 200mA 1IF  (Synology DiskStation 650047CBB9343987)
  2-1:1.0         (IF) 08:06:50 2EPs () usb-storage host8 (synoboot)
|__usb3          1d6b:0001:0310 09  1.10   12MBit/s 0mA 1IF  (uhci_hcd 0000:00:1a.0)
 3-0:1.0          (IF) 09:00:00 1EP  () hub
|__usb4          1d6b:0001:0310 09  1.10   12MBit/s 0mA 1IF  (uhci_hcd 0000:00:1d.0)
 4-0:1.0          (IF) 09:00:00 1EP  () hub
|__usb5          1d6b:0001:0310 09  1.10   12MBit/s 0mA 1IF  (uhci_hcd 0000:00:1d.1)
 5-0:1.0          (IF) 09:00:00 1EP  () hub
|__usb6          1d6b:0001:0310 09  1.10   12MBit/s 0mA 1IF  (uhci_hcd 0000:00:1d.2)
 6-0:1.0          (IF) 09:00:00 1EP  () hub
|__usb7          1d6b:0002:0310 09  2.00  480MBit/s 0mA 1IF  (Linux 3.10.108 etxhci_hcd-170202 Etron xHCI Host Controller 0000:04:00.0)
 7-0:1.0          (IF) 09:00:00 1EP  () ethub
  |__7-1         1a86:55d4:0442 02  1.10   12MBit/s 132mA 2IFs (ITEAD SONOFF Zigbee 3.0 USB Dongle Plus V2 20220811090153)
  7-1:1.0         (IF) 02:02:01 1EP  () cdc_acm tty/ttyACM1
  7-1:1.1         (IF) 0a:00:00 2EPs () cdc_acm
|__usb8          1d6b:0003:0310 09  3.00 5000MBit/s 0mA 1IF  (Linux 3.10.108 etxhci_hcd-170202 Etron xHCI Host Controller 0000:04:00.0)
 8-0:1.0          (IF) 09:00:00 1EP  () ethub
root@DiskStation:~#

fruitvlieg1, 2022-11-30:
root@DiskStation:~# ls /dev/ttyA*
/dev/ttyACM1
root@DiskStation:~#

robertklep, 2022-11-30:
Oh great, now it's `/dev/ACM1` 😭 

In any case though, the device is supported without additional drivers. Like I said, I don't use ZHA in Home Assistant so I can't really help you on what the correct configuration should be 🥴

fruitvlieg1, 2022-11-30:
Thanks for your support Robert, Have a nice day!

stanczakp, 2023-01-10:
Hello,

I'm also trying to connect the sonoff dongle to my synology nas, but for now without any success.

@fruitvlieg1 have finally got it workin? Did you installed on of the drivers?

@robertklep is it maybe possible to add support for a smart card reader? It is detected as
` 4-0:1.0          (IF) 09:00:00 1EP  () hub
  |__4-2         0403:6001:0600 00  2.00   12MBit/s 90mA 1IF  (FTDI FT232R USB UART AI068S9P)`
but no tty device is created.

robertklep, 2023-01-10:
@stanczakp I would first have to know which (Linux) driver should be used for this device, and then see if it's possible to compile it for DSM. Do you have another Linux machine on which you can test this device to see which driver(s) it requires?

stanczakp, 2023-01-10:
@robertklep yes, I have a debian machine, where this device is working. What info should I provide to you?

robertklep, 2023-01-10:
@stanczakp when you plug the device into your machine, check `dmesg`, it should show which driver is loaded for it. Can you create a new issue for this? So we don't bother @fruitvlieg1 with off-topic comments 😅