robertklep/dsm7-usb-serial-drivers #51: Building for rtd1296 7.1

alt219, 2022-12-13:
Just FYI, had to use the following to build for rtd1296 7.1:

`sudo /toolkit/pkgscripts-ng/PkgCreate.py --build-opt='-J' -v 7.1 --min-sdk 7.1 4.4.x`

For the record, I've attached both the failing build log and the successful build log from /toolkit/build_env/ds.rtd1296-7.1/logs.

[4.4.x.build.failure.txt](https://github.com/robertklep/dsm7-usb-serial-drivers/files/10213794/4.4.x.build.failure.txt)
[4.4.x.build.success.txt](https://github.com/robertklep/dsm7-usb-serial-drivers/files/10213795/4.4.x.build.success.txt)

robertklep, 2022-12-13:
Yes, this is a very common issue that I also get all the time, the build process just isn't very stable. Sometimes it works, sometimes it doesn't (with different errors, too).

A fix that seems to work for me is to edit the file `/usr/local/x86_64-pc-linux-gnu/x86_64-pc-linux-gnu/sys-root/usr/lib/modules/DSM-7.1/build/scripts/Makefile.build` in `/toolkit/builld_env/ds.PLATFORM-7.1` and modify the rules `rule_cc_o_c` en `rule_as_o_S` so they don't remove the depfile.

rhjp, 2022-12-29:
@alt219 or @robertklep , are you able to add the builds of these modules to a branch in this or a forked repo?

Thanks

robertklep, 2022-12-29:
@rhjp see https://github.com/robertklep/dsm7-usb-serial-drivers/tree/main/modules/rtd1296/dsm-7.1

rhjp, 2022-12-30:
Thanks @robertklep , they work great

robertklep, 2022-12-31:
Thanks for the feedback!

closed 2022-12-31 by robertklep