robertklep/dsm7-usb-serial-drivers #52: CP210x driver / kernel module needed for DSM 6.2.4-25556 Update 6, kernel 3.10.105 (avoton architecture)

twobeseen, 2022-12-15:
I know that most people search for driver for DSM 7 but I still have DSM 6.2.4 installed.

I tried the "SynoKernel USB Serial driver" from https://synocommunity.com/package/synokernel-usbserial for my DSM 6.2.4-25556 Update 6, kernel 3.10.105 (avoton architecture) but with no luck:

I installed the package SynoKernel USB Serial drivers and
wanted to load kernel module "cp210x.ko" in my DS1515+ (avoton) with DSM 6.2.4-25556 Update 6, kernel 3.10.105
but get the dmesg message:

cp210x: version magic '3.10.102 SMP mod_unload ' should be '3.10.105 SMP mod_unload '

Anybody could perhaps help and compile the package/kernel module with the right kernel version please?

Thanks

robertklep, 2022-12-16:
As far as I can remember, DSM6 came with most of the serial drivers already installed in `/lib/modules`

twobeseen, 2022-12-16:
usbserial.ko is present, but no cp210x.ko ...

robertklep, 2022-12-16:
Yes, you do need cp210x.ko for it.

Perhaps you can find the original Jadahl drivers here: https://web.archive.org/web/20220630112631/http://www.jadahl.com/drivers_6.2/

twobeseen, 2022-12-16:
Yeah, would be nice to try that, sadly the archive didn't archive the DS1515+ (= avoton) one *cry* ;-)
(The Wayback Machine has not archived that URL.)

robertklep, 2022-12-16:
Ah too bad 😭 I can only help with DSM7 drivers, perhaps someone that's still running DSM6 and has downloaded the drivers previously might be able to help you (I assume you want to run some sort of home automation platform, so perhaps on a forum specific to that).

twobeseen, 2023-01-07:
I could successfully compile the cp210x.ko kernel module myself today. I am quite happy :-)

I used the following info for this:
https://www.bydavy.com/2012/01/compiling-linux-kernel-modules-for-synology-devices/
https://hallard.me/how-to-install-kernel-modules-on-synology-ds1010-dsm-4-1/

I downloaded the correct Synology toolchain "avoton-gcc493_glibc220_linaro_x86_64-GPL.txz" (more or less the right gcc compiler) from here:
https://sourceforge.net/projects/dsgpl/files/Tool%20Chain/DSM%206.2.4%20Tool%20Chains/Intel%20x86%20Linux%203.10.105%20%28Avoton%29/    (i found the link to sourceforge in this PDF: https://global.download.synology.com/download/Document/Software/DeveloperGuide/Firmware/DSM/6.0/enu/DSM_Developer_Guide_6_0.pdf)

Additionally I downloaded the Synology NAS GPL Sources that are needed to compile CP210x driver:
https://global.download.synology.com/download/ToolChain/Synology%20NAS%20GPL%20Source/6.2-25556/avoton/linux-3.10.x.txz

Like in the info documents described i did the following things on my virtual Ubuntu):
Already I had necessary devel packages, compiler etc. already installed in Ubuntu but needed ncurses package:
`sudo apt-get install ncurses-dev
sudo tar -xf /path-to/avoton-gcc493_glibc220_linaro_x86_64-GPL.txz -C /usr/local
sudo tar -xf /path-to/linux-3.10.x.txz -C /usr/local/x86_64-pc-linux-gnu/
sudo cd /usr/local/x86_64-pc-linux-gnu/linux-3.10.x
sudo cp ./synoconfigs/avoton .config
sudo make ARCH=x86 CROSS_COMPILE=/usr/local/x86_64-pc-linux-gnu/bin/x86_64-pc-linux-gnu- oldconfig
sudo make ARCH=x86 CROSS_COMPILE=/usr/local/x86_64-pc-linux-gnu/bin/x86_64-pc-linux-gnu- menuconfig`

In the menuconfig i could enable to build the cp210x driver under...

After enabling to compile the cp210x i then compiled all modules (too lazy to just let compile this one module for cp210x):
`sudo make ARCH=x86 CROSS_COMPILE=/usr/local/x86_64-pc-linux-gnu/bin/x86_64-pc-linux-gnu- modules`

The successful built module cp210x.ko could be found then under:
`ls -Al /usr/local/x86_64-pc-linux-gnu/linux-3.10.x/driver/usb/serial/cp210x.ko`

closed 2023-01-07 by twobeseen

robertklep, 2023-01-07:
Thanks for the elaborate explanation!