robertklep/dsm7-usb-serial-drivers #53: Invalid module format on Geminilake

Semme, 2023-01-07:
Hey there, 

If I run insmod ch341.ko or other drivers, it tells me "insmod: ERROR: could not insert module ch341.ko: Invalid module format".

My System: DS 720+ with Sonoff Zigbee 3.0 Dongle
Kernel: Linux ds720 4.4.180+ #42962 SMP Tue Oct 18 15:07:03 CST 2022 x86_64 GNU/Linux synology_geminilake_720+

I have downloaded the latest DSM 7.1 Geminilake modules from here. Does anybody know this issue? 

Regards,
Simon

robertklep, 2023-01-07:
Are you sure you downloaded the actual .ko file correctly? There are some download instructions [here](https://github.com/robertklep/dsm7-usb-serial-drivers#downloading-a-module).

Semme, 2023-01-07:
Yes, I am sure. I manually copied the link from github to the terminal. Then downloaded it with wget directly to the right folder. Just did it again to be sure:

[image]

Btw, this is my lsusb. Seems like the stick is recognized, but there isnt a serial port for it.

`|__usb1          1d6b:0002:0404 09  2.00  480MBit/s 0mA 1IF  (Linux 4.4.180+ xhci-hcd xHCI Host Controller 0000:00:15.0) hub
  |__1-2         10c4:ea60:0100 00  1.10   12MBit/s 100mA 1IF  (Silicon Labs Sonoff Zigbee 3.0 USB Dongle Plus 0001)
  |__1-4         f400:f400:0100 00  2.00  480MBit/s 200mA 1IF  (Synology DiskStation 7F00120D97345A45)
|__usb2          1d6b:0003:0404 09  3.00 5000MBit/s 0mA 1IF  (Linux 4.4.180+ xhci-hcd xHCI Host Controller 0000:00:15.0) hub`

robertklep, 2023-01-07:
That's not the correct URL, you're downloading the HTML page 😅 The correct URL is this: https://github.com/robertklep/dsm7-usb-serial-drivers/raw/main/modules/geminilake/dsm-7.1/ch341.ko

Semme, 2023-01-07:
Oh wow, you are totally right. That was embarrassing. Now its working just fine. Thank you, Robert!

robertklep, 2023-01-07:
No problem, you're not the first 😉 Glad to hear it works!

closed 2023-01-07 by robertklep