robertklep/dsm7-usb-serial-drivers #54: Smart card reader

stanczakp, 2023-01-10:
Hello @robertklep 

as requested in the previous topic I created new one and here is the dmeg output:

`
[63622.973161] usb 1-3: USB disconnect, device number 13
[63622.973357] ftdi_sio ttyUSB1: error from flowcontrol urb
[63622.973562] ftdi_sio ttyUSB1: FTDI USB Serial Device converter now disconnected from ttyUSB1
[63622.973610] ftdi_sio 1-3:1.0: device disconnected
[63627.623708] usb 1-3: new full-speed USB device number 14 using xhci_hcd
[63627.751780] usb 1-3: device descriptor read/64, error -71
[63628.017397] usb 1-3: New USB device found, idVendor=0403, idProduct=6001, bcdDevice= 6.00
[63628.017404] usb 1-3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[63628.017408] usb 1-3: Product: FT232R USB UART
[63628.017411] usb 1-3: Manufacturer: FTDI
[63628.017414] usb 1-3: SerialNumber: AI068S9P
[63628.020604] ftdi_sio 1-3:1.0: FTDI USB Serial Device converter detected
[63628.020704] usb 1-3: Detected FT232RL
[63628.020953] usb 1-3: FTDI USB Serial Device converter now attached to ttyUSB0
`

robertklep, 2023-01-10:
Thanks, it looks like the required driver is `ftdi_sio`.

What's the package arch of your Synology? Look here: https://kb.synology.com/en-uk/DSM/tutorial/What_kind_of_CPU_does_my_NAS_have

Also, which kernel version does your Synology run? `uname -a`

stanczakp, 2023-01-10:
I have `Cedarview` and `Linux DSS22 3.10.108 #42962 SMP Tue Oct 18 15:04:45 CST 2022 x86_64 GNU/Linux synology_cedarview_713+`

stanczakp, 2023-01-10:
Wow, that was fast. I have installed the module, but when I try to run it in docker I get this error message:
`ERROR: IO_Serial_GetStatus: ioctl(TIOCMGET): Bad file descriptor`
does it ring a bell?

robertklep, 2023-01-10:
No, sorry :(

Try plugging in the device on your Debian machine and see if there are any dependencies that it requires:
```
lsmod | grep ftdi_sio
```

stanczakp, 2023-01-10:
It returns only this
```
lsmod | grep ftdi_sio
ftdi_sio               57344  1
usbserial              53248  3 ftdi_sio
usbcore               299008  7 ftdi_sio,usbserial,xhci_hcd,cdc_acm,usb_storage,xhci_pci,uas
```

robertklep, 2023-01-10:
I wonder if it may have to do with Docker. Are you running the container as privileged?

stanczakp, 2023-01-10:
Yes, it is privileged, but it looks like a problem with access to serial devices from docker on synology...

robertklep, 2023-01-10:
Not in general, I used various serial devices (Z-Wave/Zigbee dongles) from Docker when I was still using my Synology NAS.

stanczakp, 2023-01-10:
I was able right now to run the snoff zigbe2mqtt dongle with this commands

```
/sbin/modprobe usbserial
/sbin/modprobe ftdi_sio
/sbin/modprobe cdc-acm
```

and it works in deed. I will make further investigation

stanczakp, 2023-01-11:
Everything works fine. Thank you for the support.

closed 2023-01-11 by stanczakp