robertklep/dsm7-usb-serial-drivers #55: USB tuner Realtek RTL2838 DVB-T

florian987, 2023-01-16:
Hello,

I am trying to get my USB tuner working on my Synology NAS. I first tried the drivers provided in this repo but no luck.

Using my laptop, I found out the needed driver was this one:
`dvb-usb-rtl28xxu.ko`

I found an article explaining how to cross-compile drivers, and tried using these toolchain and source:
* https://global.download.synology.com/download/ToolChain/Synology%20NAS%20GPL%20Source/7.0-41890/apollolake/linux-4.4.x.txz
* https://global.download.synology.com/download/ToolChain/toolchain/7.1-42661/Intel%20x86%20Linux%204.4.180%20%28Apollolake%29/apollolake-gcc850_glibc226_x86_64-GPL.txz

But it is still not working, here is the dmesg outpout, after doing an insmod:
`[49083.649640] dvb_usb_rtl28xxu: version magic '4.4.180 SMP mod_unload ' should be '4.4.180+ SMP mod_unload '`

Seems the linux version is not correct, but cannot find the right one on Synology.

Here is my NAS version:
`Linux naz 4.4.180+ #42962 SMP Wed Sep 21 10:56:47 CST 2022 x86_64 GNU/Linux synology_apollolake_218+`

Any help on getting this working would be greatly appreciated!

Thanks

robertklep, 2023-01-16:
Not sure why the version magic is different, I've never come across that issue myself.

But aside from that, my guess is that you're going to need more modules than just `dvb-usb-rtl28xxu.ko` to get it working (like `dvb_usb_core` and possibly others). You can check which dependencies the module has on your laptop: 
```
lsmod | grep rtl28xxu
```

FWIW, this repo only contains USB serial drivers.

florian987, 2023-01-18:
Thanks for you hints, I managed to cross compile this driver!

bfg100k, 2023-03-02:
Hi @florian987, did you manage to get the tuner working? I'm also after the same thing. Are you able to share some tips on how you managed to compile the driver and install it in DSM 7.1?

florian987, 2023-03-02:
Not yet, I solved the issue concerning magic numbers, forcing them with an env variable during compilation.

For cross-compilation I used this article (it's in French, use translate):
https://benjamin-balet.info/multimedia/synology/compiler-un-driver-pour-le-synology/

I am willing to give more details, or even the compiled drivers, but not sure this repo is the right place. I understand they have serial drivers only. But searching on Github, I could not find any other initiatives for our use case.

Regarding the tuner driver itself, it has a lot of dependencies, and I am kind of stuck right now (and reaching my knowledge limit regarding these topics), so if you have any hints, it would be greatly appreciated!

IamMattM, 2023-05-27:
@florian987 : I would be interested to try out your compiled "dvb-usb-rtl28xxu.ko" for DSM7 if it was possible to share it ?

Mikei386, 2023-09-09:
I would be very very interested as well to get my RTL USB Dongle to work. 
Would you like to share your driver or at least a guid how exactly you have done it?

Thank you very much.

Koky05, 2023-09-11:
For RTL drivers look into: [bb-qq](https://github.com/bb-qq/r8152)

Mikei386, 2023-09-11:
> 

Thanks for your suggestion. The device mentioned here is a DVB-T dongle, not an Ethernet device.
So I think it’s not the correct solution for this instance. (Or am I missing something?)