robertklep/dsm7-usb-serial-drivers #56: PL2303 for Bromolow

szazketto, 2023-01-18:
Hi Robert,

I am still running DSM 6.0.2 (kernel 3.10.77) and would like to use a PL2303 usb-serial converter. Can you help with the correct files?
Thanks

robertklep, 2023-01-18:
The SynoCommunity package list contains Bromolow drivers which should work on DSM 6: https://synocommunity.com/package/synokernel-usbserial

robertklep, 2023-01-18:
By the way, you don't need to run the SynoCommunity package manager if you want to install those drivers. If you download the package for your platform, you can extract the files from it manually. The `.spk` file is a TAR archive, so if you rename it with the extension `.tar` most unarchivers will happily extract it.

One of the extracted files is called `package.tgz` which contains the actual kernel modules. I just hope that they will work since they are targeting a different kernel version than yours.

szazketto, 2023-01-18:
Thanks for the hint. Unfortunately I get this error:

[3720858.998572] pl2303: version magic '3.10.102 SMP mod_unload ' should be '3.10.77 SMP mod_unload 

It seems the different kernel version is the problem.
Do you happen to know if there is a version for 3.10.77 somewhere?

robertklep, 2023-01-18:
I looked but can't find anything 😢

szazketto, 2023-01-18:
OK, thanks. I will try to upgrade my old xpenology to 6.1 or 6.2.