robertklep/dsm7-usb-serial-drivers #57: Drivers for Braswell not working on Synology DS 216+ (Sonoff Dongle Plus)

mariusrga, 2023-01-20:
I've tried to use the drivers for my Synology NAS (Braswell), but they do not work. Getting below error:

insmod: ERROR: could not insert module /lib/modules/cp210x.ko: Invalid module format

Appreciate if you could help me with this.

Thank you.

robertklep, 2023-01-20:
Hi! You didn't download the correct file(s), please follow the instructions [here](https://github.com/robertklep/dsm7-usb-serial-drivers#downloading-a-module).

mariusrga, 2023-01-20:
Mostly yes.
1. Downloaded the files from here: https://github.com/robertklep/dsm7-usb-serial-drivers/tree/main/modules/braswell
2. Copied all the files to /lib/modules
3. tried to use insmod to insert them (ex: sudo insmod /lib/modules/cp210x.ko), but was getting those errors for each file.
4. usb-serial-drivers.sh does not exist in the Braswell folder, so not sure what is it.
Thank you.

robertklep, 2023-01-20:
Mostly, but not entirely, because you didn't download the actual `.ko` files, otherwise you wouldn't get that particular error.

mariusrga, 2023-01-20:
My platform: Linux HomeNAS 3.10.108 #42962 SMP Tue Oct 18 15:05:36 CST 2022 x86_64 GNU/Linux synology_braswell_216+

mariusrga, 2023-01-20:
> Mostly, but not entirely, because you didn't download the actual `.ko` files, otherwise you wouldn't get that particular error. The file `usb-serial-drivers.sh` isn't in the Braswell folder, it's in the [top-level folder](https://github.com/robertklep/dsm7-usb-serial-drivers).

These are the files I downloaded, for Braswell. If these are not correct, then, which are please? https://github.com/robertklep/dsm7-usb-serial-drivers/tree/main/modules/braswell/dsm-7.1

robertklep, 2023-01-20:
Click (_don't_ download) each `.ko` file and press the Download button, as explained in the download instructions that I linked to.

[image]

mariusrga, 2023-01-20:
I used sudo wget then the link to each file, for ex;
sudo wget https://github.com/robertklep/dsm7-usb-serial-drivers/blob/main/modules/braswell/dsm-7.1/ch341.ko

Is this not OK?

robertklep, 2023-01-20:
No, that's not okay, the correct link is https://github.com/robertklep/dsm7-usb-serial-drivers/raw/main/modules/braswell/dsm-7.1/ch341.ko