robertklep/dsm7-usb-serial-drivers #58: build from source instructions for dsm7.1

annegerben, 2023-01-20:
I assume I also need to include some source files.
I googled but need some hints for ti_usb3410_5052.
Where can I find these, can someone point me in the right direction.

robertklep, 2023-01-20:
What exactly is it you want to do?

annegerben, 2023-01-20:
I want to use usb-ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus on ds218+ on dsm7.1.
ti_usb3410_5052.ko is not available for dsm 7.1 for [appollolake ](https://github.com/robertklep/dsm7-usb-serial-drivers/tree/main/modules/apollolake/dsm-7.1) and the import of 7.0 fails.
So I want to build it.

when I run `sudo /toolkit/pkgscripts-ng/PkgCreate.py -X 0 -P 1 -v 7.0 --min-sdk 7.0 -p apollolake 4.4.x`
it fails. But I assume it need to put something related to ti_usb3410_5052 in the 4.4.x folder?

robertklep, 2023-01-20:
I'm pretty sure the Sonoff Dongle Plus can use the `ch341` driver.

robertklep, 2023-01-20:
In case you still need the TI driver, you can find it [here](https://github.com/robertklep/dsm7-usb-serial-drivers/blob/main/modules/apollolake/dsm-7.1/ti_usb_3410_5052.ko).

annegerben, 2023-01-20:
Thank you for the above, I try with ch341 first and the ti_usb3410_5052.
But still if I would like to compile (and help with the project) I need some hints.

this is what I get:
`sudo /home/ag/toolkit/pkgscripts-ng/PkgCreate.py -X 0 -P 1 -v 7.1 --min-sdk 7.1 -p apollolake 4.4.x                                                                                                      [sudo] password for ag:
============================================================
                   Parse argument result
------------------------------------------------------------
platforms     : apollolake
env_section   : default
env_version   : 7.1
dep_level     : 1
parallel_proj : 2
branch        : master
suffix        :
collect       : False
collecter     : True
link          : True
update_link   : False
build         : True
install       : True
only_install  : False
parallel      : 1
build_opt     :  --parallel 2
install_opt   :  --parallel 2
print_log     : False
tee           : True
sdk_ver       : 7.1
package       : 4.4.x

Processing [7.1-42784]: apollolake
============================================================
              Start to run "Traverse project"
------------------------------------------------------------
Projects: 4.4.x

============================================================
                Start to run "Link Project"
------------------------------------------------------------
Link /home/ag/toolkit/pkgscripts-ng -> /home/ag/toolkit/build_env/ds.apollolake-7.1/pkgscripts-ng
Link /home/ag/toolkit/source/4.4.x -> /home/ag/toolkit/build_env/ds.apollolake-7.1/source/4.4.x

============================================================
                Start to run "Build Package"
------------------------------------------------------------
[apollolake] env PackageName=4.4.x /pkgscripts-ng/SynoBuild --apollolake -c --min-sdk 7.1 --parallel 2 4.4.x
Set cache size limit to 8.0 Gbytes
Statistics cleared
[INFO] projectList=4.4.x
[INFO] Start to build 4.4.x.
[Error] There is no build scripts for 4.4.x
[Error] Build project fail!
----------------- Time cost statistics -----------------
Time cost: 00:00:00 [Build-->4.4.x]
[INFO] Build 4.4.x finished!
Time cost: 00:00:00 [Build-->4.4.x]
########################################################
      Error(s) occurred on project "4.4.x"
########################################################
1 projects, 1 failed, 0 blocked.
[Error] Check [/logs/error.build] for fixing errors.
Traceback (most recent call last):
  File "/home/ag/toolkit/pkgscripts-ng/include/python/exec_env.py", line 76, in execute
    output = commandrunner.run(cmd, display=display, **kwargs)
  File "/home/ag/toolkit/pkgscripts-ng/include/python/commandrunner.py", line 39, in run
    raise RunShellFailed(p.returncode, cmd, output)
commandrunner.RunShellFailed

During handling of the above exception, another exception occurred:

[... 25 lines cut ...]
    env.execute(cmd, display=len(self.env_config.platforms)
  File "/home/ag/toolkit/pkgscripts-ng/include/python/exec_env.py", line 79, in execute
    raise EnvError(
exec_env.EnvError: ('Execute chroot /home/ag/toolkit/build_env/ds.apollolake-7.1 env PackageName=4.4.x /pkgscripts-ng/SynoBuild --apollolake -c --min-sdk 7.1 --parallel 2 4.4.x failed', 'Set cache size limit to 8.0 Gbytes\nStatistics cleared\n[INFO] projectList=4.4.x\n[INFO] Start to build 4.4.x.\n[Error] There is no build scripts for 4.4.x\n[Error] Build project fail!\n\n----------------- Time cost statistics -----------------\nTime cost: 00:00:00 [Build-->4.4.x]\n[INFO] Build 4.4.x finished!\nTime cost: 00:00:00 [Build-->4.4.x]\n\n########################################################\n      Error(s) occurred on project "4.4.x"\n########################################################\n1 projects, 1 failed, 0 blocked.\n\n[Error] Check [/logs/error.build] for fixing errors.')
"""

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/home/ag/toolkit/pkgscripts-ng/PkgCreate.py", line 184, in <module>
    main(sys.argv[1:])
  File "/home/ag/toolkit/pkgscripts-ng/PkgCreate.py", line 177, in main
    packer.pack_package()
  File "/home/ag/toolkit/pkgscripts-ng/include/python/pkguniform.py", line 437, in pack_package
    worker.execute()
  File "/home/ag/toolkit/pkgscripts-ng/include/python/pkgcommon.py", line 86, in execute
    self._process_output(self._run(*argv))
  File "/home/ag/toolkit/pkgscripts-ng/include/python/pkgcommon.py", line 212, in _run
    return doPlatformParallel(self.run_command, self.env_config.platforms)
  File "/home/ag/toolkit/pkgscripts-ng/include/python/parallel.py", line 60, in doPlatformParallel
    output[item] = results[item].get()
  File "/usr/lib/python3.10/multiprocessing/pool.py", line 774, in get
    raise self._value
exec_env.EnvError: ('Execute chroot /home/ag/toolkit/build_env/ds.apollolake-7.1 env PackageName=4.4.x /pkgscripts-ng/SynoBuild --apollolake -c --min-sdk 7.1 --parallel 2 4.4.x failed', 'Set cache size limit to 8.0 Gbytes\nStatistics cleared\n[INFO] projectList=4.4.x\n[INFO] Start to build 4.4.x.\n[Error] There is no build scripts for 4.4.x\n[Error] Build project fail!\n\n----------------- Time cost statistics -----------------\nTime cost: 00:00:00 [Build-->4.4.x]\n[INFO] Build 4.4.x finished!\nTime cost: 00:00:00 [Build-->4.4.x]\n\n########################################################\n      Error(s) occurred on project "4.4.x"\n########################################################\n1 projects, 1 failed, 0 blocked.\n\n[Error] Check [/logs/error.build] for fixing errors.')
`

robertklep, 2023-01-20:
This is the command I use to build:
```
sudo /toolkit/pkgscripts-ng/PkgCreate.py -c -I -v 7.1 --min-sdk 7.1 -p apollolake 4.4.x
```

However, the build process is very unstable and regularly fails (and sometimes it doesn't 🤷🏼‍♂️).

In my experience, it helps when the depfiles in `rule_cc_o_c` and `rule_as_o_S` aren't removed. Those build rules can be found in this file:
```
/toolkit/build_env/ds.*-7.1/usr/local/x86_64-pc-linux-gnu/x86_64-pc-linux-gnu/sys-root/usr/lib/modules/DSM-7.1/build/scripts/Makefile.build
```

annegerben, 2023-01-20:
I'll move the toolkit from home to root and give it another try

annegerben, 2023-01-26:
Not sure what is wrong with my DS218+ but all module give error "Unknown symbol in module" while running insmod.
To not be dependent on these modules anymore (and with future dsm update) I'm now using a VM and usb passthrough.

Thanks for your help.

closed 2023-01-26 by annegerben