robertklep/dsm7-usb-serial-drivers #59: Tead Sonoff Zigbee 3.0 USB Dongle Plus V2 model "ZBDongle-E" driver for geminilake architecture

RobMcRob24, 2023-01-22:
Hi, 

is there a possibility for this driver too?
I need it for my DS920+, where the ch341 driver didn't work correctly.

Thanks!

robertklep, 2023-01-22:
What does "didn't work correctly" mean?

RobMcRob24, 2023-01-22:
The device is recognized, I can see /dev/ttyACM0, but the recognition in a docker container for zigbee2mqtt is not working.
There I only get messages like:

Zigbee2MQTT:info  2023-01-21 17:57:16: Starting Zigbee2MQTT version 1.29.2 (commit #bb3e8f6)
Zigbee2MQTT:info  2023-01-21 17:57:16: Starting zigbee-herdsman (0.14.83-hotfix.0)
Zigbee2MQTT:error 2023-01-21 17:57:37: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2023-01-21 17:57:37: Failed to start zigbee
Zigbee2MQTT:error 2023-01-21 17:57:37: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
Zigbee2MQTT:error 2023-01-21 17:57:37: Exiting...
Zigbee2MQTT:error 2023-01-21 17:57:37: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
    at ZStackAdapter.start (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:103:27)
    at Controller.start (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:132:29)
    at Zigbee.start (/app/lib/zigbee.ts:58:27)
    at Controller.start (/app/lib/controller.ts:101:27)
    at start (/app/index.js:107:5)

robertklep, 2023-01-22:
If the device entry is created, the driver is working properly.

For these dongles, you typically have to use a USB extension cord, especially if you use a USB 3.0 port.

Also, how did you configure the Docker container?

RobMcRob24, 2023-01-22:
Docker container via portainer:

version: '3.8'
services:
  zigbee2mqtt:
    container_name: zigbee2mqtt
    image: koenkk/zigbee2mqtt:${TAG-latest}
    user: "1026:100"    
    restart: ${RST-unless-stopped}
    networks:
      zigbee2mqtt_net:
       ipv4_address: 192.168.24.202
    privileged: true
    volumes:
      - /volume2/docker/zigbee2mqtt/config:/app/data
    ports:
      # Frontend port
      - 8080:8080
    environment:
      - TZ=${TZ-Europe/Berlin}
    labels:
      - com.centurylinklabs.watchtower.enable=${AutoUpdate-true}
      - hide_container=false
    devices:
      - /dev/ttyACM0:/dev/ttyUSB0

networks:
 zigbee2mqtt_net:
  external: true
  name: syn_macvlan

Up to now no USB extension tried. Will try it this morning.

RobMcRob24, 2023-01-22:
Tried also with
- /dev/ttyACM0:/dev/ttyACM0

Both didn't work

RobMcRob24, 2023-01-22:
Unfortunately still the same with an USB extension cable:

Using '/app/data' as data directory
Zigbee2MQTT:info  2023-01-22 09:33:52: Logging to console and directory: '/app/data/log/2023-01-22.09-33-52' filename: log.txt
Zigbee2MQTT:info  2023-01-22 09:33:52: Starting Zigbee2MQTT version 1.29.2 (commit #bb3e8f6)
Zigbee2MQTT:info  2023-01-22 09:33:52: Starting zigbee-herdsman (0.14.83-hotfix.0)
Zigbee2MQTT:error 2023-01-22 09:34:13: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2023-01-22 09:34:13: Failed to start zigbee
Zigbee2MQTT:error 2023-01-22 09:34:13: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
Zigbee2MQTT:error 2023-01-22 09:34:13: Exiting...
Zigbee2MQTT:error 2023-01-22 09:34:13: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
    at ZStackAdapter.start (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:103:27)
    at Controller.start (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:132:29)
    at Zigbee.start (/app/lib/zigbee.ts:58:27)
    at Controller.start (/app/lib/controller.ts:101:27)
    at start (/app/index.js:107:5)

And from my ds920+:

roland@ds922:~$ lsusb -iI
|__usb1          1d6b:0002:0404 09  2.00  480MBit/s 0mA 1IF  (Linux 4.4.180+ xhci-hcd xHCI Host Controller 0000:00:15.0)
 1-0:1.0          (IF) 09:00:00 1EP  () hub
  |__1-2         1a86:55d4:0442 02  1.10   12MBit/s 132mA 2IFs (ITEAD SONOFF Zigbee 3.0 USB Dongle Plus V2 20220816180351)
  1-2:1.0         (IF) 02:02:01 1EP  () cdc_acm tty/ttyACM0
  1-2:1.1         (IF) 0a:00:00 2EPs () cdc_acm
  |__1-4         f400:f400:0100 00  2.00  480MBit/s 200mA 1IF  (Synology Diskstation 7F001B0DA9A06B95)
  1-4:1.0         (IF) 08:06:50 2EPs () usb-storage host6 (synoboot)
|__usb2          1d6b:0003:0404 09  3.00 5000MBit/s 0mA 1IF  (Linux 4.4.180+ xhci-hcd xHCI Host Controller 0000:00:15.0)
 2-0:1.0          (IF) 09:00:00 1EP  () hub
roland@ds922:~$ ls -l /dev/ttyACM0
crwxrwxrwx 1 root root 166, 0 Jan 22 09:27 /dev/ttyACM0

robertklep, 2023-01-22:
> at ZStackAdapter.start 

Ah, it's trying to use the zStack driver, which isn't going to work with ZBDongle-E, you need to use the `ezsp` driver for that. See [the manual](https://www.zigbee2mqtt.io/guide/adapters/#experimental).

RobMcRob24, 2023-01-22:
Thank you very much!
It's running now!

Zigbee2MQTT:info  2023-01-22 09:47:20: Starting Zigbee2MQTT version 1.29.2 (commit #bb3e8f6)
Zigbee2MQTT:info  2023-01-22 09:47:20: Starting zigbee-herdsman (0.14.83-hotfix.0)
Zigbee2MQTT:info  2023-01-22 09:47:23: zigbee-herdsman started (resumed)
Zigbee2MQTT:info  2023-01-22 09:47:23: Coordinator firmware version: '{"meta":{"maintrel":"3 ","majorrel":"6","minorrel":"10","product":8,"revision":"6.10.3.0 build 297"},"type":"EZSP v8"}'
Zigbee2MQTT:info  2023-01-22 09:47:23: Currently 0 devices are joined:

closed 2023-01-22 by RobMcRob24