robertklep/dsm7-usb-serial-drivers #60: (Solved) Sonoff Zigbee 3.0 USB Dongle Plus V2 model "ZBDongle-E"

Flo040, 2023-01-25:
This problem is solved but since i got really frustrated I will just post it to inform others of a possible (simple) hardware problem.

The ZBDongle-E with and without USB extender **did** work on my DS2013 but **did not** work on my DS2020+. A new USB extender cable was al that was needed to get it work. (with the driver ch341 installed according to Robert's description off course)

------
I've tried to get my Sonoff Zigbee 3.0 USB Dongle Plus V2 model "ZBDongle-E" to work for over 3 weeks but no matter what I tried and I couldn't get it to work.

- Sonoff Zigbee 3.0 USB Dongle Plus ZBDongle-E
- Synology DS220+
- DSM: DSM 7.1.1-42962 Update 1

Because I got desperate, I also picked up my old Synology 2013 to see if the ZBDongle-E works on it.
DSM: DSM 6.2.4-25556 Update 6

There the dongle did work. After a new install of my DS2020+ and freshly installed drivers it still didn't work. I have tried a number of things with various tips from Robert and other tutorials on the web but still no ttyACM or ttyUSB1 or whatever. While I try to describe my problem in as much detail as possible for this page i wanted to be sure I exlcluded all possible own mistakes and hardware problems so i decided to buy a new extender cable. Just to be sure...... first try..... BINGO!

Robert thanks for the support!!

romanr, 2023-02-16:
I'm having somewhat similar error. Before you added correct extender did USB Dongle show up in OS, or it was completely not present among devices?

ivanopa1, 2025-04-11:
> I'm having somewhat similar error. Before you added correct extender did USB Dongle show up in OS, or it was completely not present among devices?

Same issue with me. I've solved it by installing manually package with USB drivers manually (from UsbSerialDrivers_apollolake-6.2_6-7.spk) found on the Domoticz forum). I have still DSM 6.2   After that I was glad to see  /dev/ttyUSB0  appearing in devices...