robertklep/dsm7-usb-serial-drivers #62: Can't setup /dev/ttyACM0 for Sonoff Zigbee 3.0 on DSM 7.1

Chacsam, 2023-02-06:
Hello,
I have tried the whole setup many times, but I always end up at the same stage with my configuration: ttyUSB0 is setup, but not ttyACM0, which I seem to need for zigbee2MQTT...

My config:
Synology DS920+ with DSM 7.1.1-42962 Update 3
so I went for modules/geminilake/dsm-7.1   (+ ti_usb_3410_5052.ko from the dsm-7.0 folder)

The devices
```
lsusb
|__usb1          1d6b:0002:0404 09  2.00  480MBit/s 0mA 1IF  (Linux 4.4.180+ xhci-hcd xHCI Host Controller 0000:00:15.0) hub
  |__1-1         10c4:ea60:0100 00  2.00   12MBit/s 100mA 1IF  (ITead Sonoff Zigbee 3.0 USB Dongle Plus 32733906b9c9eb11983a8b4f1d69213e)
  |__1-2         051d:0002:0106 00  2.00   12MBit/s 100mA 1IF  (American Power Conversion Back-UPS BX950MI  FW:295201G -302201G  9B2120A18510)
  |__1-4         f400:f400:0100 00  2.00  480MBit/s 200mA 1IF  (Synology DiskStation 7F00095D8213F246)
```
and
```
dmesg |grep tty
[    0.000000] Command line: root=/dev/md0 intel_iommu=igfx_off netif_num=2 HddEnableDynamicPower=1 syno_hw_version=DS920+ vender_format_version=2 synoboot2 sn=20A0SBR2T043X macs=001132de8451,001132de8452 syno_ttyS0=pciserial,0x0:0x18.0x2,115200 syno_ttyS1=pciserial,0x0:0x18.0x0,115200
[    0.000000] Kernel command line: root=/dev/md0 intel_iommu=igfx_off netif_num=2 HddEnableDynamicPower=1 syno_hw_version=DS920+ vender_format_version=2 synoboot2 sn=20A0SBR2T043X macs=001132de8451,001132de8452 syno_ttyS0=pciserial,0x0:0x18.0x2,115200 syno_ttyS1=pciserial,0x0:0x18.0x0,115200
[    4.782954] dw-apb-uart.1: ttyS1 at MMIO 0xa1415000 (irq = 54, base_baud = 115200) is a 16550A
[    4.803814] console [ttyS0] disabled
[    4.807676] dw-apb-uart.2: ttyS0 at MMIO 0xa1417000 (irq = 6, base_baud = 115200) is a 16550A
[    4.817380] console [ttyS0] enabled
[   18.250763] synobios open /dev/ttyS1 success
[   29.300766] systemd[1]: Created slice system-serial\x2dgetty.slice.
[   29.307781] systemd[1]: Starting system-serial\x2dgetty.slice.
[   31.581130] synobios open /dev/ttyS1 success
[ 1373.395038] usb 1-1.1: cp210x converter now attached to ttyUSB0
[ 2106.083721] cp210x ttyUSB0: cp210x converter now disconnected from ttyUSB0
[ 2161.174399] usb 1-1: cp210x converter now attached to ttyUSB0
```
so, I desperately end up with
```
ls /dev/ttyA*
ls: cannot access '/dev/ttyA*': No such file or directory
```
and a unhappy zigbee2MQTT docker.

Am I missing something?

robertklep, 2023-02-06:
Just configure z2m to use `/dev/ttyUSB0` instead, it should work fine.

Chacsam, 2023-02-06:
Why did I miss that one?
I just had to map it in my docker compose...
Thanks!

closed 2023-02-06 by Chacsam