robertklep/dsm7-usb-serial-drivers #64: Version Magic error on denverton, 7.1, Zigbee 3.0 V2

romanr, 2023-02-16:
Having this error:
`kernel: [] ti_usb_3410_5052: version magic '3.10.108 SMP mod_unload ' should be '4.4.180+ SMP mod_unload '`

The software on DiskStation detects the USB device correctly, it shows up, but when it tries to load/modprobe it fails.

romanr, 2023-02-16:
Hi and thanks!

I followed steps from README:
```sh
git clone https://github.com/robertklep/dsm7-usb-serial-drivers.git
cp dsm7-usb-serial-drivers/modules/denverton/dsm-7.1/* /lib/modules
cp dsm7-usb-serial-drivers/usb-serial-drivers.sh /usr/local/etc/rc.d
chmod +x /usr/local/etc/rc.d/usb-serial-drivers.sh
/usr/local/etc/rc.d/usb-serial-drivers.sh start
```

Do I need to remove other drivers? It tries to load them all and fails?

robertklep, 2023-02-16:
I think you initially copied the wrong modules so there's now a `/lib/modules/ti_usb_3410_5052.ko` for the wrong architecture that's causing an error. Just remove it.

FWIW, even though you're getting that error, the rest of the modules should have loaded successfully.

romanr, 2023-02-16:
Thank you for reply!

after removing `/lib/modules/ti_usb_3410_5052.ko` and running the script nothing happens. no error, no error log or any other message. Homeassistant says same as before - "module probe failed", nothing in its logs or system logs.

Is there a way to test the USB device somehow?

robertklep, 2023-02-16:
Please consider that I have no idea what you're trying to do and how you're doing it 😊

Are you running Home Assistant in a Docker container, or in a virtual machine?

romanr, 2023-02-16:
Oh it's nothing special, HomeAssistant as plain DSM OS package. I
Home assistant  detects "Zigbee 3.0 v2" and prompts to install it. But then it fails with `Failed to probe the usb device` without any errors in log.
This is its page dedicated to Zigbee USB: https://www.home-assistant.io/integrations/zha

robertklep, 2023-02-17:
What does `lsusb -cui` show? From what I remember from ZHA, you have to explicitly pass it a device path; which path are you using for it?

romanr, 2023-02-17:
Thank you for helping , I appreciate :) 
here it is:
```
|__usb1          1d6b:0002:0000 09  2.00  480MBit/s 0mA 1IF  (Linux 4.4.180+ xhci-hcd xHCI Host Controller 0000:00:15.0) hub
  |__1-1         f400:f400:0100 00  2.00  480MBit/s 200mA 1IF  (Synology DiskStation 7F001AEB2313F233)
  1-1:1.0         (IF) 08:06:50 2EPs () usb-storage host6 (synoboot)
  |__1-2         0764:0600 00  2.00   12MBit/s 2mA 1IF  (CPS OR1500ERM1U-AU GBULX2000211)
  1-2:1.0         (IF) 03:00:00 2EPs () usbfs
  |__1-3         1a86:55d4:0000 02  1.10   12MBit/s 132mA 2IFs (ITEAD SONOFF Zigbee 3.0 USB Dongle Plus V2 20220818080600)
  1-3:1.0         (IF) 02:02:01 1EP  () cdc_acm tty/ttyACM0
 ```

robertklep, 2023-02-17:
So your device is there, as `/dev/ttyACM0`.

romanr, 2023-02-17:
amateur question - if device present in /dev means driver is loaded successfully? and in this case `/dev/tty/ttyACM0` can be added to software configuration, etc.?

robertklep, 2023-02-17:
That's correct, yes.

romanr, 2023-02-17:
I found out what it is, as with 9 out of 10 problems in linux it's permissions!: `Permission denied: '/dev/ttyACM0'`
googled it and it mostly advices about adding user to "dialout" group. also this issue https://github.com/Koenkk/zigbee2mqtt/issues/12218

I'll keep looking if homeassistant user need to be added to some group.
Thanks a lot for taking time and your help! I can make MR with Readme update about troubleshooting, in case someone else have same problems.

romanr, 2023-02-19:
`chmod` didn't help. 
Finally got it fixed by running:
`chown sc-homeassistant /dev/ttyACM0`

closed 2023-02-19 by romanr