robertklep/dsm7-usb-serial-drivers #66: Error running usb-serial-drivers.sh

crispy78, 2023-02-27:
When running usb-serial-drivers.sh I got an error mentioned below. Reason is that I've copy/pasted it from github and used notepad to create a file. The error was:
```
root@root:/lib/modules# /usr/local/etc/rc.d/usb-serial-drivers.sh start
-ash: /usr/local/etc/rc.d/usb-serial-drivers.sh: /bin/sh^M: bad interpreter: No such file or directory
```
I've solved it by running, where the "sed" command replaces spurious CR characters:
```
chown root:root /lib/modules/ch341.ko
chown root:root /lib/modules/cp210x.ko
chown root:root /lib/modules/pl2303.ko
chown root:root /lib/modules/rndis_host.ko
chown root:root /lib/modules/ti_usb_3410_5052.ko
sed -i -e 's/\r$//' /usr/local/etc/rc.d/usb-serial-drivers.sh
```
Hope this will help others.

GravityRZ, 2023-03-22:
maybe if you first click the raw button and after that copy the script to a file you will not get those nasty characters.
just my 2 cents :-)

crispy78, 2023-08-06:
It's probably a bit late, but I used the RAW-source. After a few Synology updates I'm downloading the zip-file extract the needed files to my NAS, copy it to the desired directory, CHOWN the files and run the .sh.