robertklep/dsm7-usb-serial-drivers #68: 'MAC transaction expired' (240)

lluisd, 2023-03-04:
Aqara roller shade companion E1 device stops working. It was working at least during 3 days and now without doing anything it reports errors and also happens again at least 2 times more. I use sonoff zigbee as coordinator in a synology ds920+ 

I am suspecting this issue with 2 factors:
- The synology Security Advisor app which sometimes automatically commented each line with # my usb-serial-drivers.sh script but if I omit it sometimes I see that the execute permission of that file is removed.
- I also use a usb HDD dock for external backups and I don't know if this can be also related. But when I try it, it doesn't reproduce the problem.

## Device & platform
Zigbee2MQTT version: 1.30.1
Sonoff zigbee 3.0 P

## Logs
Zigbee2MQTT:error 2023-02-27 22:30:43: Publish 'set' 'position' to '0x54ef4410002e019d' failed: 'Error: Write 0x54ef4410002e019d/1 genAnalogOutput({"85":{"value":0,"type":57}}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'MAC transaction expired' (240))'
Zigbee2MQTT:info 2023-02-27 22:30:43: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'position' to '0x54ef4410002e019d' failed: 'Error: Write 0x54ef4410002e019d/1 genAnalogOutput({"85":{"value":0,"type":57}}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'MAC transaction expired' (240))'","meta":{"friendly_name":"0x54ef4410002e019d"},"type":"zigbee_publish_error"}'
Zigbee2MQTT:error 2023-02-27 23:03:27: Publish 'set' 'position' to '0x54ef4410002e019d' failed: 'Error: Write 0x54ef4410002e019d/1 genAnalogOutput({"85":{"value":0,"type":57}}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'MAC transaction expired' (240))'
Zigbee2MQTT:info 2023-02-27 23:03:27: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'position' to '0x54ef4410002e019d' failed: 'Error: Write 0x54ef4410002e019d/1 genAnalogOutput({"85":{"value":0,"type":57}}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'MAC transaction expired' (240))'","meta":{"friendly_name":"0x54ef4410002e019d"},"type":"zigbee_publish_error"}'
Zigbee2MQTT:error 2023-02-27 23:04:11: Publish 'set' 'position' to '0x54ef4410002e019d' failed: 'Error: Write 0x54ef4410002e019d/1 genAnalogOutput({"85":{"value":0,"type":57}}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'MAC transaction expired' (240))'
Zigbee2MQTT:info 2023-02-27 23:04:11: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'position' to '0x54ef4410002e019d' failed: 'Error: Write 0x54ef4410002e019d/1 genAnalogOutput({"85":{"value":0,"type":57}}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'MAC transaction expired' (240))'","meta":{"friendly_name":"0x54ef4410002e019d"},"type":"zigbee_publish_error"}'
Zigbee2MQTT:info 2023-02-27 23:04:27: Configuring '0x54ef4410002e019d'
Zigbee2MQTT:error 2023-02-27 23:04:55: Publish 'set' 'position' to '0x54ef4410002e019d' failed: 'Error: Write 0x54ef4410002e019d/1 genAnalogOutput({"85":{"value":80,"type":57}}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'MAC transaction expired' (240))'
Zigbee2MQTT:info 2023-02-27 23:04:55: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'position' to '0x54ef4410002e019d' failed: 'Error: Write 0x54ef4410002e019d/1 genAnalogOutput({"85":{"value":80,"type":57}}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'MAC transaction expired' (240))'","meta":{"friendly_name":"0x54ef4410002e019d"},"type":"zigbee_publish_error"}'
Zigbee2MQTT:error 2023-02-27 23:05:39: Failed to configure '0x54ef4410002e019d', attempt 1 (Error: Read 0x54ef4410002e019d/1 aqaraOpple([1034], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":4447,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'MAC transaction expired' (240))
at ZStackAdapter.sendZclFrameToEndpointInternal (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:415:23)
at Queue.executeNext (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:32:32))
Zigbee2MQTT:info 2023-02-27 23:05:39: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/configure', payload '{"data":{"id":"0x54ef4410002e019d"},"error":"Failed to configure (Read 0x54ef4410002e019d/1 aqaraOpple([1034], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":4447,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'MAC transaction expired' (240)))","status":"error","transaction":"vpxe4-1"}'
Zigbee2MQTT:error 2023-02-27 23:06:23: Publish 'set' 'state' to '0x54ef4410002e019d' failed: 'Error: Write 0x54ef4410002e019d/1 genAnalogOutput({"85":{"value":0,"type":57}}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'MAC transaction expired' (240))'
Zigbee2MQTT:info 2023-02-27 23:06:23: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'state' to '0x54ef4410002e019d' failed: 'Error: Write 0x54ef4410002e019d/1 genAnalogOutput({"85":{"value":0,"type":57}}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'MAC transaction expired' (240))'","meta":{"friendly_name":"0x54ef4410002e019d"},"type":"zigbee_publish_error"}'

robertklep, 2023-03-04:
I don't see how this can be related to the serial drivers. If they don't get loaded properly, z2m won't start up because it'll be unable to find the serial device.

Any other issues are beyond the scope of what I can do for you, other than advise that you have to make sure you use an USB extension lead the connect the Sonoff to your Synology.

lluisd, 2023-03-04:
> I don't see how this can be related to the serial drivers. If they don't get loaded properly, z2m won't start up because it'll be unable to find the serial device.
>
> Any other issues are beyond the [...]

I opened the same issue in Koenkk/zigbee2mqtt but I am sure they will say that this is because I am running it on the Synology.

Drivers are load properly and it's working for few days after zigbee devices stop being detected. I never reboot the NAS and /dev/ttyUSB0 still appears. I will try to use an extension usb to try things.

I want to detail that when that happens the Sonoff dongle is detected and when I unplug it the zigbee2mqtt also detects that I did that and shows a message of unpluged device. The last time this happened to me the only solution was to stop the my HDD dock, because it wasn't mounted but it was powered on, I switch off the power and then magically my zigbee device works again. Without doing that I was also unable to pair, /dev/ttyUSB0 was shown and i run again the .sh script but Zigbee2Mqtt doesn't detect the pairing device anymore until I power off my HDD.

Peharpas someone has similar problem, that's the reason I also write it here because all of use we use Synology with some similar sonoff device.

robertklep, 2023-03-04:
If you think it may be related to your HDD dock, it could very well be an interference issue. USB3 is notorious for causing interference on 2.4Ghz, which can especially affect Zigbee devices, being a low-powered protocol.

Make sure you have a very good shielded cable for your HDD dock, and also use a long (1m minimum) extension lead to your Sonoff dongle. And if possible, use the front USB port of your NAS for the dongle.

lluisd, 2023-03-04:
what do you think now?

[image]

robertklep, 2023-03-04:
Does it work better now?