robertklep/dsm7-usb-serial-drivers #71: DS718+ under DSM 7.1 USB serial install not working

boggiz, 2023-03-12:
**Linux NAS-DS718 4.4.180+ #42962 SMP Fri Feb 10 16:32:29 CST 2023 x86_64 GNU/Linux  synology_apollolake_718+**

Hello,
I just bought a new NAS to do Docker with Domoticz and I can't install the Apollolake USB drivers provided by Robert Klep.

This_ is what I got when restarting the NAS:
insmod: ERROR: could not insert module /volume3/Originaux/DSMusbserial71/ch341.ko: Unknown symbol in module
insmod: ERROR: could not insert module /volume3/Originaux/DSMusbserial71/cp210x.ko: Unknown symbol in module
insmod: ERROR: could not insert module /volume3/Originaux/DSMusbserial71/pl2303.ko: Unknown symbol in module
insmod: ERROR: could not insert module /volume3/Originaux/DSMusbserial71/ti_usb_3410_5052.ko: Unknown symbol in module
insmod: ERROR: could not insert module /volume3/Originaux/DSMusbserial71/rndis_host.ko: Unknown symbol in _module

Thank's for your help

robertklep, 2023-03-12:
How are you loading the modules? I think that `usbserial.ko` isn't being loaded, which is a requirements for the others.

boggiz, 2023-03-12:
Thank's for your response.
Usbserial.ko is in lib/modules like there:
[image]
I tested 2 ways:
1) Created a task root with insmod command 
2) with putty with insmod too
With the same error !

robertklep, 2023-03-12:
But did you _load_ `usbserial.ko` using `insmod` before loading the other modules?
```
insmod /lib/modules/usbserial.ko
insmod /volume3/Originaux/DSMusbserial71/ch341.ko
```

boggiz, 2023-03-12:
insmod USBserial must be the first of the list?
Is there an order to respect for the other modules?
I try it ...

GravityRZ, 2023-03-12:
would using modprobe not solve this issue
it should also load rthe dependant module (usbserial)

found this in anoyther thread

modprobe is an intelligent command, it looks for dependencies while loading a module. Suppose, if I loaded a module, which has symbols defined in some other module (this module path is given inside the main module). So, modprobe loads the main module and the dependent module.

But if insmod is used, it won't load the dependency, and hence it will give compilation errors like Unresolved symbols. In this case, we have to manually look for dependent module and need to load them in order to resolve the errors.

boggiz, 2023-03-12:
With your help the USB drivers were OK excepted LPM
[image]
But my RFXtrx433XL on domoticz is not OK which drivers should I add ?

boggiz, 2023-03-12:
Oups! I do not read your previous message.

GravityRZ, 2023-03-12:
i am still on dsm 6.2 but also have a 718+

the rfxcom uses ftdi_sio

try this (or use the insmod alternative)
modprobe usbserial
modprobe ftdi_sio
modprobe cdc-acm

robertklep, 2023-03-13:
> would using modprobe not solve this issue

@GravityRZ I've never gotten `modprobe` to work with these modules and never bothered to fully investigate 😬

GravityRZ, 2023-03-13:
glad you got it working.

out of interest, how are these loaded when you reboot.
do you use a bootup script 
/usr/local/etc/rc.d/usb-serial-drivers.sh start

robertklep, 2023-03-13:
@GravityRZ from a task scheduler job, see [this comment](https://github.com/robertklep/dsm7-usb-serial-drivers/issues/71#issuecomment-1465215418)

GravityRZ, 2023-03-13:
Oops. did not read that. thanks

boggiz, 2023-03-13:
I think I rejoiced too soon!
This morning Domoticz was working but I stopped it for the day.
Tonight after reboot, no more USB . Why all seems OK ?
[image]

robertklep, 2023-03-13:
@boggiz that sounds like an issue I can't help with, the drivers are seemingly loaded just fine.

GravityRZ, 2023-03-13:
if you look in /dev
do you see the device files
ttyACM0 (used by zwave stick)
ttyUSB0 (used by rfxcom)

if not unplug the devices and plug them back in.

if you see them now look at the rights of those files
mine have root:root rights with default 0600
ttyACM0 has 0666 rights

i see that some people give them 0777 rights.
maybe that will help when rebooting

boggiz, 2023-03-13:
With or without RFX connection no files ttyACM0 or ttyUSB0 in /dev. 
Where should I find them?

boggiz, 2023-03-13:
[image]
How to reload modules 

After the 'USB registered drivers' I found these errors :
[image]

GravityRZ, 2023-03-13:
do you run your docker containers with high privileges?  if not please do
those driver files should appear when you install the driver(either modprobe or insmod) and something connects to those ports.

GravityRZ, 2023-03-13:
if things still do not work
disable that script that runs the drivders at startup

reboot

after reboot load the drivers manually

insmod /lib/modules/usbserial.ko
insmod /lib/modules/ftdi_sio.ko
insmod /lib/modules/cdc-arm.ko
insmod /volume3/Originals/DSMusbserial71/ch341.ko
insmod /volume3/Originals/DSMusbserial71/cp210x.ko
insmod /volume3/Originals/DSMusbserial71/pl2303.ko
insmod /volume3/Originals/DSMusbserial71/ti_usb_3410_5052.ko

after this unplug and plug the devices

robertklep, 2023-03-13:
> After the 'USB registered drivers' I found these errors :

If you're using a USB extension lead with your device, try replacing it.
If you're _not_ using a USB extension load: try adding one.

boggiz, 2023-03-14:
Today I tried all the tests you suggested with insmod or modprobe in scheduled or manual mode:
I have never seen the file /dev/ttyUSB0

Below is the latest task execution report:
[image]

I even tried SynoCli Kernel Tools without result.

For information, only the following programs are installed on the NAS:
- DSM 7.1.1-42962 update 4
- Docker
- Domoticz in a docker container

I don't understand it worked once!
What more ?

robertklep, 2023-03-14:
To be honest, I don't understand why you're making it difficult for yourself. If you don't want to use the provided shell script, that's perfectly fine, but at least take some pointers from it. It's really a very simple script that works for pretty much everyone.

It doesn't make sense trying to run `mknod /dev/usb/ttyUSB0`, and the reason it's failing is because `/dev/usb/` simply doesn't exist on DSM.

You also seem to be loading _just_ `usbserial.ko` and `ftdi_sio.ko` and not any of the other drivers, which is likely the cause for `/dev/ttyUSB*` not appearing: you're just not loading the correct drivers for your device.

GravityRZ, 2023-03-14:
also you only need to load the drivers for your devices, not all of them.

usbserial is mandatory for some other drivers(dependency)
ftdi_sio.ko is for the rfxcom
cdc_acm is for the zwave stick

what other serial devices are you using

also did you check out you containers. are they running with high privileges
your container need to be able to find those ports outside the container
[image]

robertklep, 2023-03-14:
In English:
[image]

boggiz, 2023-03-14:
[image]

@[robertklep](https://github.com/robertklep)  I known that but it's more faster to see if the task was executed well than to go through Putty!

@GravityRZ  I used only 1 RFXtrx433XL and YES I used High privilege in scheluler and manual sudo under Putty

boggiz, 2023-03-14:
With manual sudo all modules were mounted without error but without ttyUSB0 !

robertklep, 2023-03-14:
Is the log still showing errors? If so, there's something wrong with the connection between your devices and your NAS.

GravityRZ, 2023-03-14:
ok.
RFXtrx433XL uses ftdi_sio
insmod /lib/modules/ftdi_sio.ko
this should genertate /dev/ttyUSB0 when the device is connected.

what usb port do you use, i use those on the back
DO NOT USE A USBHUB, only direct connect.
disconnect the cable and reconnect(on both sides, also the side of the rfx)

GravityRZ, 2023-03-14:
also hopefully you did not try to flash the RFX with new firmware  under domoticz
if you did this might be the problem

boggiz, 2023-03-14:
No I did not flash my RFX it is in version 10.44 and it works perfectly on my DS215+ on DSM 6.2.4. 
Tomorrow I will install the same DSM 7.1 on another disk without anything else to redo the tests.

GravityRZ, 2023-03-15:
ok. good luck. we are puzzled why it is not working

robertklep, 2023-03-16:
If the device requires physically reconnecting it a number of times, I don't see how you can do that in software. It seems to me that the issue is with the device, not your NAS (or possibly a bad connector/cable but I assume you ruled that out).

GravityRZ, 2023-03-17:
i also would suspect a bad connection on the rfxtrx or perhaps the cable
it is easy to open so maybe inspect the connector and the connector solder points

or ask on the rfxtrx site if this is a familiar problem.

i had an inssue with mine wgere the p1 functionionality was not working.
it turned out i needed a lower pull up resistor.
for all we know this is a power problem
mine is connected to the top usb port on the back

boggiz, 2023-03-17:
Thanks for the leads, I'll take a look. 
But why does my RFX work 100% under DSM 6.2.4?

GravityRZ, 2023-03-17:
if you have it NOW working on 6.2.4 that is indeed strange

i would focus on only using the ftdi_sio driver because that one is used for the rfx on 6.2.4 (and probably also on 7.1)

did you try this solution(offered by other user)

//create directory /etc/modules-load.d

//goto that directory and create file 99usbdrivers.conf
//put this in that file

usbserial.ko
ftdi_sio.ko
cdc-acm.ko

//copy those files to
/lib/modules/modules.dep

//Create depmod as symlink to kmod
ln -s /usr/bin/kmod /sbin/depmod

//Execute depmod, no param's needed, ignore warning and error if any
depmod
// Restart
systemctl restart systemd-modules-load

robertklep, 2023-03-17:
> But why does my RFX work 100% under DSM 6.2.4?

It's also running on another NAS, isn't it?

boggiz, 2023-03-17:
@robertklep Yes but not at the same time. For Domoticz I have got only 1 RFXcom for 2 syno.

GravityRZ, 2023-03-17:
i think what robert means is

NAS-A 6.2.4 RFXcom works
NAS-B(718+) 7.1:RFXcom does not work

is this the case or did you tet on 1 nas system.

eg. 
718+ 6.2.4 RFX working
718+ 7.1 RFX not working
back to
718+ 6.2.4 RFX working again.

please tell exactly what you did to rtule out things

boggiz, 2023-03-17:
I absolutely agree with you. 
This is why I would like to downgrade my new DS718+ to DSM 6.2.4 but is very complicated.
@GravityRZ can you send me the link. I will try this script

GravityRZ, 2023-03-17:
it is not a script. those are manual commands.
are you able to log in to the file system with root rights (eg with ssh, winscp)
if not you can try this procedure beginning at step 7(adding a new package source for downloading)
it looks like you can add usb serial package(just as in 6.2.4) 
not sure if this will stick after reboot

[https://mariushosting.com/synology-how-to-add-usb-support-on-dsm-7/]

GravityRZ, 2023-03-17:
downgrading is not difficult if you have root access

if you are not familiar with this i suggest you look into this
absolute control of the synology is key when you want to change things.

google this
ssh access to synology

also look at this tool winscp.
if configured correctly it is really simple to login your synology, execute scripts, copy files or transfer files from you network share/pc to your synology

https://www.reddit.com/r/synology/comments/qjci2i/downgrade_synology_dsm_7_to_6/

GravityRZ, 2023-03-25:
i noticed that touching the usb connectors sometime results in loss of connection.
i think this is because the synology connector is USB3 and the cable is USB2

so changing the cable to  a better type might help you