robertklep/dsm7-usb-serial-drivers #72: USB modules for Armada370 on DSM 7.1

tcyna, 2023-03-19:
Hi, 
Can you compile USB modules for DS115j DSM 7.1?  It's on Armada370.

Thanks!

robertklep, 2023-03-20:
Try these: https://github.com/robertklep/dsm7-usb-serial-drivers/tree/main/modules/armada370/dsm-7.1

tcyna, 2023-03-20:
Thank you very much! Unfortunately not working. I loaded the modules but nothing happened. No new device appeared like e.g. /dev/ttyUSB0.

robertklep, 2023-03-21:
If you loaded the modules and literally nothing happened, the modules loaded just fine, so the issue isn't with the modules per se.

What's the device you're trying to get support for? What does `/usr/syno/bin/lsusb -cui` show when the device is plugged in?

tcyna, 2023-03-21:
I'm trying to add CC2531 ZigBee module. On DSM 6.2 worked pretty well.

Output of lsusb -cui:

|__usb2          1d6b:0002:0302 09  2.00  480MBit/s   0mA 1IF  (Linux 3.2.101 ehci_hcd Marvell Orion EHCI ehci_marvell.1) hub
  |__2-1         0451:16a8:0009 02  2.00   12MBit/s  50mA 2IFs (Texas Instruments TI CC2531 USB CDC __0X00124B001CD6EB4C)`
  2-1:1.0         (IF) 02:02:01 1EP  ()
  2-1:1.1         (IF) 0a:00:00 2EPs ()

robertklep, 2023-03-21:
`CDC` suggests it needs the `cdc-acm.ko` driver, which is already present on DSM:
```
sudo insmod /lib/modules/usbserial.ko
sudo insmod /lib/modules/cdc-acm.ko
```
 It will also like provide `/dev/ttyACM0`, not `/dev/ttyUSB0`.

tcyna, 2023-03-21:
This is strange. I sure I had /dev/ttyUSB0 on DSM 6.2 and adding new device in Home Assistant was as simple as choosing that device. 

Now I have:
|__usb2          1d6b:0002:0302 09  2.00  480MBit/s   0mA 1IF  (Linux 3.2.101 ehci_hcd Marvell Orion EHCI ehci_marvell.1) hub
  |__2-1         0451:16a8:0009 02  2.00   12MBit/s  50mA 2IFs (Texas Instruments TI CC2531 USB CDC __0X00124B001CD6EB4C)
  2-1:1.0         (IF) 02:02:01 1EP  () cdc_acm tty/ttyACM0
  2-1:1.1         (IF) 0a:00:00 2EPs () cdc_acm

But when I try to add /dev/ttyACM0 I've get "Failed to connect"