robertklep/dsm7-usb-serial-drivers #75: Perfectly working in 7.1.1.42962

GravityRZ, 2023-03-24:
No problems, just some info for all the DSM 7 users

i finally upgraded from 6.2.4 to 7 because support on 6.2.4. will end in june.
I made a backup of all the important stuff(photo's, video's, containers)

i used the default button to upgrade to 7.0 (did not want to risk uploading to a later version and break things)
i am only using usbserial.ko  ftdi_sio.ko(for RFX) and cdc-acm.ko(for zwave gen5 stick) so i modified the script to only include those
these files are default present so no need to download them.
After the upgrade i used winscp and copied the script to /usr/local/etc/rc.d
i logged in with putty
chmod +x /usr/local/etc/rc.d/usb-serial-drivers.sh
/usr/local/etc/rc.d/usb-serial-drivers.sh start

i made a copy of the default drivers

everything is working smoothly so i can confirm the prodecure which @robertklep made is working perfectly
rebooted and things still work so drivers start correctly at bootup.

after cleaning up obsolete packages(i removed oznu MQTT package which deleted the complete container, oops)
i upgraded to the latest 7.1.1.42962 build, rebooted, everything still working
Because i only use the default drivers i can upgrade without breaking things(until synology screws up)
i again made a copy of the default drivers

Thanks Robert

Jenda987, 2023-07-02:
Be careful with using this /usr/lib/modules-load.d/95-usb-serial.conf in DSM 7.2. It causes me lot of troubles and DSM reinstalls. Once I added this it wasn't possible to complete boot up of DSM 7.2 ...

GravityRZ, 2023-07-02:
thanks

did you upgrade from 7.1.1 to 7.2 or straight from 6.4 to 7.2?

also do you now use the old procedure /usr/local/etc/rc.d/usb-serial-drivers.sh?

this might be something to look into because it seems that synology is again making our life miserable by blocking usb drivers

> Be careful with using this /usr/lib/modules-load.d/95-usb-serial.conf in DSM 7.2. It causes me lot of troubles and DSM reinstalls. Once I added this it wasn't possible to complete boot up of DSM 7.2 ...

Jenda987, 2023-07-02:
> thanks
> 
> did you upgrade from 7.1.1 to 7.2 or straight from 6.4 to 7.2? also do you now use the old procedure /usr/local/etc/rc.d/usb-serial-drivers.sh?
> 
> last question, when you look into the /usr/lib/modules-load.d directory what do you see.
> 
> the file name 95-usb-serial was choosen because the default file names have a lower number. eg those file are started with lowest number first so the last file(our usb file) needs to have the highest number. if this is not the case(eg standard files have a number higher than 95) then we might get problems
> 
> this might be something to look into because it seems that synology is again making our life miserable by blocking usb drivers
> 
> > Be careful with using this /usr/lib/modules-load.d/95-usb-serial.conf in DSM 7.2. It causes me lot of troubles and DSM reinstalls. Once I added this it wasn't possible to complete boot up of DSM 7.2 ...

I was upgrading from 7.1.1 and then I have to reset and reinstall 7.2-64570 Update 1 several times until I figured that this change is causing that my DSM was only blinking blue and then it's turned off after 10m without any web access. Default solution with /usr/local/etc/rc.d/usb-serial-drivers.sh is working without any issue.

GravityRZ, 2023-07-03:
ok. just to be sure i changed back to the old procedure also. i do not have 7.2 available as upgrade yet

Masadido, 2023-07-04:
> I was upgrading from 7.1.1 and then I have to reset and reinstall 7.2-64570 Update 1 several times until I figured that this change is causing that my DSM was only blinking blue and then it's turned [...]

Had the same issue. This procedure completely crashed my DS920+... to avoid at all cost !

Masadido, 2023-07-04:
> thanks
>
> did you upgrade from 7.1.1 to 7.2 or straight from 6.4 to 7.2? also do you now use the old procedure /usr/local/etc/rc.d/usb-serial-drivers.sh?
>
> last question, when you look into the [...]

Dear Jenda987, are you running Geminilake by any chance?

scottlinse, 2023-07-04:
> I was upgrading from 7.1.1 and then I have to reset and reinstall 7.2-64570 Update 1 several times until I figured that this change is causing that my DSM was only blinking blue and then it's turned [...]

> Had the same issue. This procedure completely crashed my DS920+... to avoid at all cost !

Where either of you able to recover DSM after doing this? I tried the Synology recovery methods for a DS920 and nothing seems to work.

Jenda987, 2023-07-05:
> Where either of you able to recover DSM after doing this? I tried the Synology recovery methods for a DS920 and nothing seems to work.
I had to use spare drive according to this https://kb.synology.com/tr-tr/DSM/tutorial/How_to_fix_hard_drive_failure_and_retrieve_the_data_from_your_hard_drives

Jenda987, 2023-07-05:
> Dear Jenda987, are you running Geminilake by any chance?

Sorry, I have V1000 (DS1621+)

Masadido, 2023-07-05:
> thanks
> did you upgrade from 7.1.1 to 7.2 or straight from 6.4 to 7.2? also do you now use the old procedure /usr/local/etc/rc.d/usb-serial-drivers.sh?
> last question, when you look into the [...]

Same here, after crash, my DS920+ behaved like it performed a hard factory reset so had to reinstall DSM 7.2 with a spare drive and then insert the drives. I hop you have good backups of the synology apps...