robertklep/dsm7-usb-serial-drivers #77: DSM 7.1 - ftdi_sio.ko for r1000

smhswiss, 2023-03-25:
Hi Robert
I would appreciate if you could build and provide the driver mentioned.
On cedarview I used your one without any issues. But the system itself got to slow and just recently replaced it with a newer DS.

I've tried to run the build myself but failed with permissioning erros that I couldn't fix.

`[2023-03-25 16:53:55,559] INFO: [r1000] env PackageName=4.4.x /pkgscripts-ng/SynoBuild --r1000 -c --min-sdk 7.1 --parallel 4 4.4.x
[2023-03-25 16:53:55,664] INFO: /bin/bash: warning: setlocale: LC_ALL: cannot change locale (en_US.utf8)
[2023-03-25 16:53:55,761] INFO: /pkgscripts-ng/include/check: line 93: /dev/null: Permission denied
[2023-03-25 16:53:55,810] INFO: ERROR: This script must be run as root
[2023-03-25 16:53:55,812] INFO: Traceback (most recent call last):
[2023-03-25 16:53:55,812] INFO:   File "/volume1/Downloads/SynoUSB_driver/pkgscripts-ng/include/python/exec_env.py", line 76, in execute
[2023-03-25 16:53:55,812] INFO:     output = commandrunner.run(cmd, display=display, **kwargs)
[2023-03-25 16:53:55,812] INFO:   File "/volume1/Downloads/SynoUSB_driver/pkgscripts-ng/include/python/commandrunner.py", line 39, in run
[2023-03-25 16:53:55,812] INFO:     raise RunShellFailed(p.returncode, cmd, output)
[2023-03-25 16:53:55,813] INFO: commandrunner.RunShellFailed
[2023-03-25 16:53:55,813] INFO: 
[2023-03-25 16:53:55,813] INFO: During handling of the above exception, another exception occurred:
[2023-03-25 16:53:55,813] INFO: 
[2023-03-25 16:53:55,813] INFO: Traceback (most recent call last):
[2023-03-25 16:53:55,813] INFO:   File "/volume1/Downloads/SynoUSB_driver/pkgscripts-ng/include/python/parallel.py", line 15, in __call__
[2023-03-25 16:53:55,813] INFO:     result = self.__callable(*args, **kwargs)
[2023-03-25 16:53:55,813] INFO:   File "/volume1/Downloads/SynoUSB_driver/pkgscripts-ng/include/python/pkgcommon.py", line 191, in run_command
[2023-03-25 16:53:55,813] INFO:     env.execute(cmd, display=len(self.env_config.platforms)
[2023-03-25 16:53:55,814] INFO:   File "/volume1/Downloads/SynoUSB_driver/pkgscripts-ng/include/python/exec_env.py", line 79, in execute
[2023-03-25 16:53:55,814] INFO:     raise EnvError(
[2023-03-25 16:53:55,814] INFO: exec_env.EnvError: ('Execute chroot /volume1/Downloads/SynoUSB_driver/build_env/ds.r1000-7.1 env PackageName=4.4.x /pkgscripts-ng/SynoBuild --r1000 -c --min-sdk 7.1 --parallel 4 4.4.x failed', '/bin/bash: warning: setlocale: LC_ALL: cannot change locale (en_US.utf8)\n/pkgscripts-ng/include/check: line 93: /dev/null: Permission denied\nERROR: This script must be run as root')
`

Or maybe you could give me a hint to resolve the build issue. The build script run's as root so, I don't get why it fails.

Thanks and regards
Stefan

robertklep, 2023-03-25:
That file should already exist as `/lib/modules/ftdi_sio.ko`

smhswiss, 2023-03-31:
> That file should already exist as `/lib/modules/ftdi_sio.ko`

Indeed - I was probably already reading to much stuff about the whole subject :-)

closed 2023-03-31 by smhswiss