robertklep/dsm7-usb-serial-drivers #78: geminilake rndis_host.ko: Unknown symbol in module

Koky05, 2023-03-31:
On new DSM 7.2 beta I clone new files from repository but during insmode I got error:

> insmod /lib/modules/rndis_host.ko
> insmod: ERROR: could not insert module /lib/modules/rndis_host.ko: Unknown symbol in module

robertklep, 2023-03-31:
Did you load `usbserial.ko` before `rndis_host.ko`?

Koky05, 2023-03-31:
Yes.

> insmod /lib/modules/usbserial.ko
> insmod: ERROR: could not insert module /lib/modules/usbserial.ko: File exists

robertklep, 2023-03-31:
Too bad, in that case it's probably required to recompile all modules.

Which platform do you have? `uname -a`

Koky05, 2023-04-01:
Linux nas 4.4.302+ #64216 SMP Tue Mar 7 17:09:39 CST 2023 x86_64 GNU/Linux synology_geminilake_720+

robertklep, 2023-04-01:
It looks like `rndis_host.ko` depends on other modules, notably `usbnet.ko` (which is available on DSM) and `mii` (which _isn't_ available on DSM). So at the moment RNDIS just isn't supported 🙁

Koky05, 2023-04-01:
Thanks for clarification.

thecifu, 2023-04-26:
I take advantage of this thread, so as not to create a new one. I have a Synology DS920+ with Geminilake Architecture and DSM7.1. I don't see compatible modules for this architecture. Could be?

Thanks in advance.

robertklep, 2023-04-26:
@thecifu see [this comment](https://github.com/robertklep/dsm7-usb-serial-drivers/issues/78#issuecomment-1492887893).