robertklep/dsm7-usb-serial-drivers #79: 7.1.1 need usb fix

pressurejets, 2023-04-16:
Linux Pressurejets 3.10.108 #42962 SMP Tue Oct 18 15:04:45 CST 2022 x86_64 GNU/Linux synology_cedarview_1513+

dsm 7.1.1 42962 update 4 

trying to get zibee 3.0 usb working in homeassisnt 

any ideas how this can be done

Lacoste313, 2023-04-16:
Same for me here with:
DSM 7.1.1-42962 Update 4
Linux SYNO-RS 3.10.108 #42962 SMP Tue Oct 18 15:00:31 CST 2022 x86_64 GNU/Linux synology_bromolow_rs3614xs+

The 7.0 driver with the script wont show /dev/ttyA* or /dev/ttyU*

robertklep, 2023-04-16:
@pressurejets which drivers did you load?

@Lacoste313 there are no Bromolow drivers for 7.1 at the moment, I'll see if I can compile those.

Can you both run `lsusb -cui` and post the results here?

Lacoste313, 2023-04-16:
> @pressurejets which drivers did you load?
> 
> @Lacoste313 there are no Bromolow drivers for 7.1 at the moment, I'll see if I can compile those.
> 
> Can you both run `lsusb -cui` and post the results here?

Hello, oh its simply for that. Thanks you for you fast reply and if you can compile it. ❤️ 

Of course, that the return:
lsusb -cui
```
|__usb1          1d6b:0002:0310 09  2.00  480MBit/s 0mA 1IF  (ehci_hcd 0000:00:1a.0) hub
  |__1-1         8087:0024:0000 09  2.00  480MBit/s 0mA 1IF  ( ffffffd1ffffffb2ffffffdbffffffad) hub
    |__1-1.1     f400:f400:0100 00  2.00  480MBit/s 200mA 1IF  (Synology DiskStation 6500432F9DB76D11)
   1-1.1:1.0      (IF) 08:06:50 2EPs () usb-storage host23 (synoboot)
|__usb2          1d6b:0002:0310 09  2.00  480MBit/s 0mA 1IF  (ehci_hcd 0000:00:1d.0) hub
  |__2-1         8087:0024:0000 09  2.00  480MBit/s 0mA 1IF  ( ffffffd1ffffffb2ffffffdbffffffa0) hub
    |__2-1.3     18d1:9302:0100 00  2.10  480MBit/s 498mA 1IF  ( ffffffd1ffffffb2ffffffdbffffffa1)
   2-1.3:1.0      (IF) ff:ff:ff 6EPs (Vendor Specific) usbfs
|__usb3          1d6b:0002:0310 09  2.00  480MBit/s 0mA 1IF  (Linux 3.10.108 etxhci_hcd-170202 Etron xHCI Host Controller 0000:0b:00.0) hub
  |__3-1         10c4:ea60:0100 00  1.10   12MBit/s 100mA 1IF  (Silicon Labs Sonoff Zigbee 3.0 USB Dongle Plus 0001)
  3-1:1.0         (IF) ff:00:00 2EPs ()
```

pressurejets, 2023-04-16:
its cedarview i believe not bromolow

pressurejets, 2023-04-16:
1usb below
|__usb1          1d6b:0002:0310 09  2.00  480MBit/s 0mA 1IF  (ehci_hcd
0000:00:1a.7) hub
|__usb2          1d6b:0002:0310 09  2.00  480MBit/s 0mA 1IF  (ehci_hcd
0000:00:1d.7) hub
  |__2-1         f400:f400:0100 00  2.00  480MBit/s 480mA 1IF  (Synology
Diskstation 0000000000000C3E)
|__usb3          1d6b:0001:0310 09  1.10   12MBit/s 0mA 1IF  (uhci_hcd
0000:00:1a.0) hub
|__usb4          1d6b:0001:0310 09  1.10   12MBit/s 0mA 1IF  (uhci_hcd
0000:00:1d.0) hub
  |__4-2         10c4:ea60:0100 00  1.10   12MBit/s 100mA 1IF  (Silicon
Labs Sonoff Zigbee 3.0 USB Dongle Plus 0001)
|__usb5          1d6b:0001:0310 09  1.10   12MBit/s 0mA 1IF  (uhci_hcd
0000:00:1d.1) hub
|__usb6          1d6b:0001:0310 09  1.10   12MBit/s 0mA 1IF  (uhci_hcd
0000:00:1d.2) hub
|__usb7          1d6b:0002:0310 09  2.00  480MBit/s 0mA 1IF  (Linux
3.10.108 etxhci_hcd-170202 Etron xHCI Host Controller 0000:04:00.0) hub
|__usb8          1d6b:0003:0310 09  3.00 5000MBit/s 0mA 1IF  (Linux
3.10.108 etxhci_hcd-170202 Etron xHCI Host Controller 0000:04:00.0) hub

robertklep, 2023-04-16:
@pressurejets yes, yours is Cedarview, @Lacoste313's is Bromolow 😊

pressurejets, 2023-04-16:
***@***.***:~$ lsusb -cui
|__usb2          1d6b:0002:0310 09  2.00  480MBit/s 0mA 1IF  (ehci_hcd
0000:00:1d.7) hub
  |__2-1         f400:f400:0100 00  2.00  480MBit/s 480mA 1IF  (Synology
Diskstation 0000000000000C3E)
  2-1:1.0         (IF) 08:06:50 2EPs () usb-storage host8 (synoboot)
|__usb4          1d6b:0001:0310 09  1.10   12MBit/s 0mA 1IF  (uhci_hcd
0000:00:1d.0) hub
  |__4-2         10c4:ea60:0100 00  1.10   12MBit/s 100mA 1IF  (Silicon
Labs Sonoff Zigbee 3.0 USB Dongle Plus 0001)
  4-2:1.0         (IF) ff:00:00 2EPs ()

pressurejets, 2023-04-16:
I'm not that clued but will help as much as I can.

robertklep, 2023-04-16:
@pressurejets it's not clear to me what exactly you need. I'm quite certain that [the current DSM 7.1 drivers for Cedarview](https://github.com/robertklep/dsm7-usb-serial-drivers/tree/main/modules/cedarview/dsm-7.1) will also work on 7.1.1. You will need to follow [the installation steps](https://github.com/robertklep/dsm7-usb-serial-drivers#installation).

pressurejets, 2023-04-16:
Thats what im trying to do but im so confused is there a easier step by
step

robertklep, 2023-04-16:
@pressurejets no sorry, there isn't, at least not from me. Perhaps search around on the Home Assistant forum to see if anyone can help, because just loading the drivers is only the start of the process.

pressurejets, 2023-04-16:
***@***.***:~$ sudo wget
https://github.com/robertklep/dsm7-usb-serial-drivers/blob/main/modules/geminilake/dsm-7.1/ch341.ko
Password:
Will not apply HSTS. The HSTS database must be a regular and
non-world-writable file.
ERROR: could not open HSTS store at '/root/.wget-hsts'. HSTS will be
disabled.
--2023-04-16 18:26:04--
https://github.com/robertklep/dsm7-usb-serial-drivers/blob/main/modules/geminilake/dsm-7.1/ch341.ko
Resolving github.com... 140.82.121.4
Connecting to github.com|140.82.121.4|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: unspecified [text/html]
Saving to: 'ch341.ko'

ch341.ko                [ <=>                ] 142.44K  --.-KB/s    in 0.09s

2023-04-16 18:26:05 (1.48 MB/s) - 'ch341.ko' saved [145855]

***@***.***:~$ sudo insmod /lib/modules/ch341.ko
insmod: ERROR: could not insert module /lib/modules/ch341.ko: Invalid
module format
***@***.***:~$

robertklep, 2023-04-16:
@pressurejets your system is based on the Cedarview platform, so you need to use the Cedarview modules, not the Geminilake modules.

Also, you're not downloading the module file itself, but its Github page. The correct URL for the Cedarview CH341 module is: https://github.com/robertklep/dsm7-usb-serial-drivers/raw/main/modules/cedarview/dsm-7.1/ch341.ko

pressurejets, 2023-04-16:
Thanks robert sorry for the noob question i just starting .
so i have wget all the right files do i need to do any think else ??

robertklep, 2023-04-16:
@Lacoste313 try these: https://github.com/robertklep/dsm7-usb-serial-drivers/tree/main/modules/bromolow/dsm-7.1

Lacoste313, 2023-04-16:
Wow, you are so speedly. I will try this now 😋

robertklep, 2023-04-16:
@pressurejets yes, follow the installation instructions 😄 especially the part about [the provided shell script](https://raw.githubusercontent.com/robertklep/dsm7-usb-serial-drivers/main/usb-serial-drivers.sh), which is necessary if you want to drivers to be loaded automatically during startup.

pressurejets, 2023-04-16:
its saving as .3
'. HSTS will be disabled.
--2023-04-16 18:40:43--
https://github.com/robertklep/dsm7-usb-serial-drivers/blob/main/modules/bromolow/dsm-7.1/ch341.ko
Resolving github.com... 140.82.121.3
Connecting to github.com|140.82.121.3|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: unspecified [text/html]
Saving to: 'ch341.ko.3'

ch341.ko.3              [ <=>                ] 137.75K  --.-KB/s    in 0.1s

2023-04-16 18:40:44 (1.40 MB/s) - 'ch341.ko.3' saved [141051]

***@***.***:~$

after i have wget files what do i do next ?

robertklep, 2023-04-16:
@pressurejets remove the previously downloaded files first, every time you download a file that already exists, `wget` will add a numerical extension to them. So `sudo rm -f /lib/modules/ch341.ko*` _then_ download it again (and same the the other drivers you want to download).

Lacoste313, 2023-04-16:
> @Lacoste313 try these: https://github.com/robertklep/dsm7-usb-serial-drivers/tree/main/modules/bromolow/dsm-7.1

That a really quick PR. All its good now with the good version.

ls /dev/ttyU*
```
/dev/ttyUSB0
```

A big thanks you. ✌️

Lacoste313, 2023-04-16:
> its saving as .3 '. HSTS will be disabled. --2023-04-16 18:40:43-- https://github.com/robertklep/dsm7-usb-serial-drivers/blob/main/modules/bromolow/dsm-7.1/ch341.ko Resolving github.com... [...]

As i can see, you dont use the good link again. U need to follow exactly what Robertlep say you here: (partiiculiary the URL, raw/main and not blop/main)

> @pressurejets your system is based on the Cedarview platform, so you need to use the Cedarview modules, not the Geminilake modules.
>
> Also, you're not downloading the module file itself, but its [...]

and removing the older files before.
when the file is deleted, just use:

`cd /lib/modules`
and
`sudo wget https://github.com/robertklep/dsm7-usb-serial-drivers/raw/main/modules/cedarview/dsm-7.1/ch341.ko`

make sure the file as been here:
`ls /lib/modules/ch341*`
If the file its here. its ok.

So, you need to make the script:
`sudo touch /usr/local/etc/rc.d/usb-serial-drivers.sh`
editing the file script with your favorite editor:
`sudo vim /usr/local/etc/rc.d/usb-serial-drivers.sh`
 put the desiring line on it:
```
#!/bin/sh
case $1 in
  start)
    insmod /lib/modules/ch341.ko > /dev/null 2>&1
    ;;
  stop)
    exit 0
    ;;
  *)
    exit 1
    ;;
`esac`
```

Make the script executable:
`sudo chmod +x /usr/local/etc/rc.d/usb-serial-drivers.sh`
and lunch it:
`sudo /usr/local/etc/rc.d/usb-serial-drivers.sh start`

Thats it. You disconnect/reconnect your zigbee key. And check with:
`ls /dev/ttyA*`
or 
`ls /dev/ttyU*`
if a result are found, your driver is correctly installed.

✌️

robertklep, 2023-04-17:
> $ ls /dev/ttyu*

Remember that Linux is case-sensistive, so lowercase "u" does not match uppercase "U".

Try this:
```
ls /dev/ttyU*
```

robertklep, 2023-04-17:
> So, you need to make the script:
> `sudo touch /usr/local/etc/rc.d/usb-serial-drivers.sh`

I would recommend using the script provided by this repository, since it will also load any dependencies that are required (`usbserial.ko`).

closed 2023-04-17 by robertklep

pressurejets, 2023-04-21:
Just done update on nas and again usb not working any ideas