robertklep/dsm7-usb-serial-drivers #80: How do you enable it once the files are copied?

alainhogue, 2023-04-19:
Hello All,

I followed all instructions and copied the .ko files for DSM 7.1.1-42962 Update 5 in /lib/modules and copied the usb-serial-drivers.sh file to /usr/local/etc/rc.d and made the file executable. Then I rebooted my DS423+. Everything looks fine, but how do I enable the UPS? 

If I go to the Control Panel/Hardware/UPS it is NOT enabled and when I try to enable it, the drop down only display the following choices none of them working:
USB UPS
SNMP UPS
Synology UPS Server

Am I missing something?

Synology DS423+
APC Smart-UPS 1000 (SU1000NET)
Serial cable with serial dongle to usb

robertklep, 2023-04-19:
If your UPS doesn't support USB directly, it's very likely not supported. Adding a USB-to-serial device in between doesn't change that, because DSM is very likely looking for a specific USB device identifier which your serial dongle doesn't match.

alainhogue, 2023-04-19:
I thought these *.ko drivers were for enabling serial to usb connectivity? Are you saying that there is no way for me to connect my old Smart-UPS using the old serial interface to my DS423+?

What is the use of this github? What people are using it for? I am confused.

robertklep, 2023-04-19:
Yes, these drivers are for serial-to-usb support, but that doesn't mean you can connect _any_ device and miraculously get it to work. In case of UPS's, DSM probably only supports devices that come with a USB port.

Besides that, according to [this page](https://networkupstools.org/protocols/apcsmart.html), APC UPS's use a non-standard serial layout, which further complicates matters.

These drivers are mostly used to support Zigbee/Z-Wave dongles.

alainhogue, 2023-04-19:
Ah! Now I understand. So I would need first to have a "converter" to reroute the signal before using my dongle? Would that work?

robertklep, 2023-04-19:
> Would that work?

I simply don't know 🥴 you can try, and if DSM recognises your UPC, great. If it doesn't, then it likely doesn't work like that. In that case, there's nothing I can do to change that.

jul974, 2025-03-21:
I made the same mistake. I bought an APC UPS. Contrary to other ones, this will not be easy to use.
Also the software Power Shut is crap. Also the cables are special (but they send it ofr free at home !!.).

And now, Robert says "UPS's use a non-standard serial layout". I am just not surpised.

I just regret to have bought it.
Anyway thanks for all that work. I learned lots of things tonight !