robertklep/dsm7-usb-serial-drivers #84: ERROR: could not insert module /lib/modules/cp210x.ko: Unknown symbol in module  synology_geminilake_920+

CrayzChris, 2023-05-01:
I updated my synology DS920+ from 7.1.1-42962 Update 5 to 7.2-64551.

The device **/dev/ttyUSB0** (Silicon Labs CP2102) is missing after the update :-(.

**Kernel**
```Linux xxx 4.4.302+ #64551 SMP Fri Apr 21 01:30:59 CST 2023 x86_64 GNU/Linux synology_geminilake_920+```

```root@xxx:/lib/modules# lsusb
|__usb1          1d6b:0002:0404 09  2.00  480MBit/s 0mA 1IF  (Linux 4.4.302+ xhci-hcd xHCI Host Controller 0000:00:15.0) hub
  |__1-1         1cf1:0030:0100 02  2.01   12MBit/s 100mA 2IFs (dresden elektronik ingenieurtechnik GmbH ConBee II DE2426406)
  |__1-2         10c4:ea60:0100 00  1.10   12MBit/s 100mA 1IF  (Silicon Labs CP2102 USB to UART Bridge Controller 0026)
  |__1-4         f400:f400:0100 00  2.00  480MBit/s 200mA 1IF  (Synology DiskStation 7F001472F413F213)
|__usb2          1d6b:0003:0404 09  3.00 5000MBit/s 0mA 1IF  (Linux 4.4.302+ xhci-hcd xHCI Host Controller 0000:00:15.0) hub
```

```
root@xxx:/lib/modules# ls -lrt
-rwx------ 1 root root   46875 May  2 00:13 usbserial.ko
-rwx------ 1 root root  272960 May  2 00:14 cp210x.ko

```

I'm not sure, can't load the modules.

```root@xxx:/lib/modules# sudo insmod /lib/modules/cp210x.ko
insmod: ERROR: could not insert module /lib/modules/cp210x.ko: Unknown symbol in module
root@xxx:/lib/modules# sudo insmod /lib/modules/usbserial.ko
insmod: ERROR: could not insert module /lib/modules/usbserial.ko: Invalid module format
```

I use the following files:
https://github.com/robertklep/dsm7-usb-serial-drivers/tree/main/modules/geminilake/dsm-7.2

Regards
Chris

robertklep, 2023-05-02:
Try this:
```
file /lib/modules/usbserial.ko
```

It sounds like you have overwritten that file with something else.

ModuCZ, 2023-05-24:
I can provide "usbserial.ko" file for DSM 7.2 (kernel 4.4.302) from DS720+ (geminilake).

schwooba, 2023-06-11:
Hi. I'm banging my head. I'm trying to load my Sonoff Zigbee USB. I can't get ttyUSB0 or ttyUSB1 to show up. I downloaded al the necessary files for my 7.2 upgrade and it worked before that time. Same kernel geminilake DS720+. When I run /usr/local/etc/rc.d/usb-serial-drivers.sh start nothing happens. Any advice you can lend would be appreciated. Thanks.

https://github.com/robertklep/dsm7-usb-serial-drivers/tree/main/modules/geminilake/dsm-7.2 

|__usb1          1d6b:0002:0404 09  2.00  480MBit/s 0mA 1IF  (Linux 4.4.302+ xhci-hcd xHCI Host Controller 0000:00:15.0) hub
  |__1-1         10c4:ea60:0100 00  2.00   12MBit/s 100mA 1IF  (ITead Sonoff Zigbee 3.0 USB Dongle Plus 6c45611e1170ec1189af2a3519c2d21c)
  1-1:1.0         (IF) ff:00:00 2EPs ()
  |__1-4         f400:f400:0100 00  2.00  480MBit/s 200mA 1IF  (Synology DiskStation 7F00072935345A59)
  1-4:1.0         (IF) 08:06:50 2EPs () usb-storage host4 (synoboot)
|__usb2          1d6b:0003:0404 09  3.00 5000MBit/s 0mA 1IF  (Linux 4.4.302+ xhci-hcd xHCI Host Controller 0000:00:15.0) hub

robertklep, 2023-06-12:
@schwooba try loading the required modules manually to see if any of them causes an error:
```
sudo insmod /lib/modules/usbserial.ko
sudo insmod /lib/modules/cdc-acm.ko
sudo insmod /lib/modules/cp210x.ko
```

robertklep, 2023-06-12:
Ah okay, that means you very likely didn't follow the [download instructions](https://github.com/robertklep/dsm7-usb-serial-drivers#downloading-a-module) properly.

The correct download URL for the required module is this: https://github.com/robertklep/dsm7-usb-serial-drivers/raw/main/modules/geminilake/dsm-7.2/cp210x.ko

schwooba, 2023-06-12:
Oh man. That worked. Thanks so much. It always comes down to the basics.

DeadlySin2, 2024-01-23:
Hello, i have the same ussue/ 
I madeeverything in the instruction by i get such error with all the files:
root@123:/bib/modules$ sudo insmod /lib/module/cp210x.ko
insmod: ERROR: could not load module /lib/module/cp210x.ko: No such file or directory
Or if i insert 
root@123:/lib/modules$ sudo file /lib/modules/cp210x.ko
/lib/modules/cp210x.ko: ELF 64-bit LSB relocatable, x86-64, version 1 (SYSV), BuildID[sha1]=37252b5255d564f271c787a7603ec5bb1aefb759, with debug_info, not stripped

robertklep, 2024-01-23:
@DeadlySin2 `modules` in the path should be plural

DeadlySin2, 2024-01-23:
Sorry for error my mistake. 
These are the commands i used to download files 
sudo wget https://github.com/robertklep/dsm7-usb-serial-drivers/raw/main/modules/geminilake/dsm-7.2/cp210x.ko
sudo wget https://github.com/robertklep/dsm7-usb-serial-drivers/raw/main/modules/geminilake/dsm-7.2/usbserial.ko
sudo wget https://github.com/robertklep/dsm7-usb-serial-drivers/raw/main/modules/geminilake/dsm-7.2/cdc-acm.ko

All files now send these errors:
root@YurlovoNAS:~# sudo insmod /lib/modules/cp210x.ko
insmod: ERROR: could not insert module /lib/modules/cp210x.ko: File exists
root@YurlovoNAS:~# sudo insmod /lib/modules/usbserial.ko
insmod: ERROR: could not insert module /lib/modules/usbserial.ko: File exists
root@YurlovoNAS:~# sudo insmod /lib/modules/cdc-acm.ko
insmod: ERROR: could not insert module /lib/modules/cdc-acm.ko: File exists

robertklep, 2024-01-23:
"file exists" means they are already loaded.

DeadlySin2, 2024-01-23:
Thanks, as i can understand the drivers are instaled and work 

root@123:/lib/modules# lsmod | grep serial
usbserial              22278  5 ch341,cp210x,ti_usb_3410_5052,pl2303,ftdi_sio
usbcore               209065  24 uas,ch341,etxhci_hcd,usblp,uhci_hcd,usb_storage,cp210x,usbserial,ti_usb_3410_5052,ehci_hcd,ehci_pci,pl2303,usbhid,ftdi_sio,cdc_acm,xhci_hcd,xhci_pci

root@123:/lib/modules# lsusb -cui
|__usb1          1d6b:0002:0404 09  2.00  480MBit/s 0mA 1IF  (Linux 4.4.302+ xhci-hcd xHCI Host Controller 0000:00:15.0) hub
  |__1-2         0bda:5411:0133 09  2.10  480MBit/s 0mA 1IF  (Generic USB2.1 Hub ffffffd6ffffffa3ffffffebffffffcb) hub
    |__1-2.1     1cf1:0030:0100 02  2.01   12MBit/s 100mA 2IFs (**dresden elektronik ingenieurtechnik GmbH ConBee II DE4354543)**
   1-2.1:1.0      (IF) 02:02:01 1EP  () usbfs
   1-2.1:1.1      (IF) 0a:00:00 2EPs () usbfs
    |__1-2.4     051d:0002:0106 00  1.10   12MBit/s 24mA 1IF  (American Power Conversion Back-UPS XS 650CI FW:892.R3 .I USB FW:R3  3B1439X26359  )
   1-2.4:1.0      (IF) 03:00:00 1EP  () usbfs
  |__1-4         f400:f400:0100 00  2.00  480MBit/s 200mA 1IF  (Synology DiskStation 6500249F9E13F269)
  1-4:1.0         (IF) 08:06:50 2EPs () usb-storage host6 (synoboot)
|__usb2          1d6b:0003:0404 09  3.00 5000MBit/s 0mA 1IF  (Linux 4.4.302+ xhci-hcd xHCI Host Controller 0000:00:15.0) hub
The Conbee II module i can see. But i cant see the device ttyACM0 or ttyUSB0 so these drivers are no working for my device?

robertklep, 2024-01-23:
I don't see the `cdc-acm.ko` module loaded, which is probably the one that's needed.

DeadlySin2, 2024-01-23:
root@123:/lib/modules# lsmod | grep serial
usbserial 22278 5 ch341,cp210x,ti_usb_3410_5052,pl2303,ftdi_sio
usbcore 209065 24 uas,ch341,etxhci_hcd,usblp,uhci_hcd,usb_storage,cp210x,usbserial,ti_usb_3410_5052,ehci_hcd,ehci_pci,pl2303,usbhid,ftdi_sio,**cdc_acm**
As i can see it is loaded, but i do not understant much. Sorry

DeadlySin2, 2024-01-23:
It is connected with the lead throw a hub. So i have DS920+ - USB HUB - Conbee2 and a USB cable from UPS
I tried to use 
sudo insmod /lib/modules/cdc-acm.ko
But it returnes insmod: ERROR: could not insert module /lib/modules/cdc-acm.ko: File exists

robertklep, 2024-01-23:
Like I said, the "File exists" error happens because the module is already loaded. Try unplugging the Conbee, then plugging it back in, then check `sudo dmesg` to see if there are any messages appearing that might point to a reason why it's not working.

ryanthuff1, 2024-04-15:
Sorry I'm having same issue, I have the correct file (raw), but it says wrong format when I load. I found this in dmesg
cp210x: version magic '4.4.302+ SMP mod_unload ' should be '4.4.180+ SMP mod_unload '

ryanthuff1, 2024-04-15:
well $hit I am still on 7.1 LOL. my bad.

robertklep, 2024-04-15:
@ryanthuff1 most modules are also available for 7.1

pacolvin, 2024-07-29:
My system is DSM is missing the cdc-acm.  Anyone know where I may be able to get it?  I can't get my Sonoff Zigbee 3 dongle to work after downloading and starting all of the other drivers.  Not sure but maybe missing the cdc-acm.ko could be the reason.

robertklep, 2024-07-29:
@pacolvin 

Can you post the output of these commands:
```
lsmod | grep serial
lsusb -cui
```

Preferably in a new issue report 👍🏻

Lou-i3, 2025-02-06:
> I can provide "usbserial.ko" file for DSM 7.2 (kernel 4.4.302) from DS720+ (geminilake).
> 
> Mine was showing same symptoms, probably overwritten by DSM 7.1 version. You can extract it from DSM 7.2 pat file after decrypting it.

@ModuCZ Can you make the "usbserial.ko" file available for download or explain how I can get it from the DSM 7.2 pat file? 
Mine isn't installing I get: 
```sh
$ insmod /lib/modules/usbserial.ko
insmod: ERROR: could not insert module /lib/modules/usbserial.ko: Invalid module format
```
I tried decrypting the pat file using https://github.com/technorabilia/syno-extract-system-patch but usbserial.ko is not in the files I get and I don't really know how to find the original. Please help. 
Thanks a lot

ModuCZ, 2025-02-06:
@Lou-i3 I'm sorry, but I have to disappoint you. Unfortunately, since my post, I have moved away from Synology and their DSM. I searched my backups, but I no longer have this data available.

Lou-i3, 2025-02-06:
@ModuCZ thanks a lot for the search!
If you remember how to get it out of the pat file, I'd love to know 😉

ModuCZ, 2025-02-06:
Here are the modules from "DSM_DS720+_72806.pat". If you have a different platform, let me know. Just provide the NAS model + current DSM version. I hope this works for you—it took me some effort to remember.

https://drive.google.com/file/d/1ivrN6MHEiJ-TOvV6dljuTXeHY2MhSLv6/view?usp=sharing

"syno-extract-system-patch" still works—don't forget to use the main version file (not the patch one). After extraction, look here:
hda1.tgz → hda1 → usr/lib/modules

Lou-i3, 2025-02-06:
Thank you so much!!!! 
I found the file where you said it was. I had looked around but not enough apparently. 
Copied it to my Synology and it worked like a charm. 
Thanks again you save me so much hassle and struggle!

ModuCZ, 2025-02-06:
Thanks for the feedback, I'm glad it helped you! :)

closed 2025-11-18 by robertklep