robertklep/dsm7-usb-serial-drivers #85: USB drivers for DS918+ DSM 7.2-64551 RC

fwiegers, 2023-05-04:
Hi,

I have a problem loading cp210x.ko for DSM 7.2-654551 RC. The same driver worked ok for the previous release (DSM 7.2-64216) worked ok without any issues.

Kernel information:
Linux Wiegers_NAS 4.4.302+ #64551 SMP Fri Apr 28 19:48:21 CST 2023 x86_64 GNU/Linux synology_apollolake_918+

I hope you can help me out.

Regards,

Frans Wiegers

robertklep, 2023-05-04:
Would be useful to know the actual problem you're having 😅

baylanger, 2023-05-04:
Please provide the output that shows the error.

fwiegers, 2023-05-04:
This is the output:

Frans@Wiegers_NAS:/lib/modules$ sudo insmod cp210x.ko
Password:
insmod: ERROR: could not insert module cp210x.ko: Invalid module format

robertklep, 2023-05-05:
@fwiegers that means you very likely didn't download the correct file. See [the instructions in the README](https://github.com/robertklep/dsm7-usb-serial-drivers#downloading-a-module).

fwiegers, 2023-05-05:
@robertklep I tried again following your exact directions. Same result
Could there be a change between 7.2-654551 RC and 7.2-64216 causing this?

robertklep, 2023-05-05:
@fwiegers what does `file /lib/modules/cp210x.ko` return?

fwiegers, 2023-05-05:
Apparently 'file' is an unknown command in DSM? 
sudo: file: command not found.
Is there a way to switch to another shell?

robertklep, 2023-05-05:
Ah sorry, I use an external package that provides `file`.

Try this:
```
tr -cd '[:print:]' < /lib/modules/cp210x.ko | head -c 50
```

fwiegers, 2023-05-05:
I get "permission denied" also when I add sudo before the command.

robertklep, 2023-05-05:
Try setting the correct permissions first: `sudo chmod 644 /lib/modules/cp210x.ko`

fwiegers, 2023-05-05:
Frans@Wiegers_NAS:/lib/modules$ sudo tr -cd '[:print:]' < /lib/modules/cp210x.ko | head -c 50
<!DOCTYPE html><html lang="en" data-color-mode="au
This means is looks like a html document, right? So wrong way of downloading?

fwiegers, 2023-05-05:
Frans@Wiegers_NAS:/lib/modules$ sudo tr -cd '[:print:]' < /lib/modules/cp210x.ko | head -c 50
<!DOCTYPE html><html lang="en" data-color-mode="au

Looks like an HTML document, right? So , wrong way of downloading?

robertklep, 2023-05-05:
No output whatsoever? That's not right.

fwiegers, 2023-05-05:
[image]
Looks like it's an HTML document? So wrong way of downloading?

robertklep, 2023-05-05:
Yes, if it's HTML you downloaded the Github page source and not the actual module.

The correct URL is this: https://github.com/robertklep/dsm7-usb-serial-drivers/raw/main/modules/apollolake/dsm-7.2/cp210x.ko

fwiegers, 2023-05-05:
Thank you very much. wget with the url, followed by the chmod, followed by insmod and now everything works!

closed 2023-05-05 by robertklep

tonnisjan, 2026-08-30:
Does this also working for the 7.4-90075

baylanger, 2026-08-30:
> Does this also working for the 7.4-90075

Yes

tonnisjan, 2026-08-31:
Thanks, what can happen in the worse case situation I mean with the next update of dsm7. Will it remove the driver? or an update is not possible anymore?