robertklep/dsm7-usb-serial-drivers #86: USB support for armada38x DSM 7.1 is not present

MySynologyDS215j, 2023-05-07:
Hello the USB support for DSM 7.1 armada38x is not present.
Could you upload it?
Will the support for USB hauppauge  wintv dualHD  be included?

robertklep, 2023-05-07:
No, TV dongles are not supported, only serial dongles. Still need them?

MySynologyDS215j, 2023-05-07:
I use it for recording free tvb by air and to see it with tablet without any payment with nas wdmycloudmirror and dvblink software. Now the wdmycloudmirror is in faseout and I would like to switch to synology.I don't know if I'm alone to do this but this don't require download line performances and the hardware if able to do it. the problem is only software. Inviata dal mio telefono-------- Messaggio originale --------Da: Robert Klep ***@***.***>Data: dom 7 mag 2023, 10:53A: robertklep/dsm7-usb-serial-drivers ***@***.***>Cc: MySynologyDS215j ***@***.***>, Author ***@***.***>Oggetto: Re: [robertklep/dsm7-usb-serial-drivers] USB support for armada38x DSM 7.1 is not present (Issue #86)
No, TV dongles are not supported, only serial dongles. Still need them?

robertklep, 2023-05-07:
The drivers in this repository only provide support for serial dongles, so you can't use them for your TV tuner I'm afraid.