robertklep/dsm7-usb-serial-drivers #87: USB-Serial not visible on DS923+ (r1000)

mgov1, 2023-05-15:
Hi,

I followed the procedure to install the drivers on my DS923+ (DSM 7.1.1-42962 Update 5).
However I cannot find the usb device.

_|__1-3         067b:23a3:0305 00  2.00   12MBit/s 100mA 1IF  (Prolific Technology Inc.  USB-Serial Controller  EKCEc19A321)_

_ls /dev | grep -i usb
usb1
usb1p1_

From what I found, the USB-Serial Controller should use the pl2303 driver.

What could be wrong?

Koky05, 2023-05-15:
DSM 7.x do not support any USB drivers, only for storage. So if you want use it via USB you need to install by your self USB driver into it.

robertklep, 2023-05-16:
@mgov1 did the pl2303 driver load without issues? What does `lsusb | grep usbserial` return?

mgov1, 2023-05-16:
Hi @robertklep, yes the driver did load without issues as far as I can see.
There is no output for "lsusb | grep usbserial".
lsusb -cui
|__usb1          1d6b:0002:0404 09  2.00  480MBit/s 0mA 1IF  (Linux 4.4.180+ xhci-hcd xHCI Host Controller 0000:06:00.3) hub
  |__1-1         f400:f400:0100 00  2.00  480MBit/s 200mA 1IF  (Synology DiskStation 650028A8202DF949)
  1-1:1.0         (IF) 08:06:50 2EPs () usb-storage host7 (synoboot)
  |__1-3         067b:23a3:0305 00  2.00   12MBit/s 100mA 1IF  (Prolific Technology Inc.  USB-Serial Controller  EKCEc19A321)
  1-3:1.0         (IF) ff:00:00 3EPs ()  
|__usb2          1d6b:0003:0404 09  3.00 5000MBit/s 0mA 1IF  (Linux 4.4.180+ xhci-hcd xHCI Host Controller 0000:06:00.3) hub
  |__2-2         0bc2:2323:0000 00  3.00 5000MBit/s 144mA 1IF  (Seagate Expansion+ NAAG7ZZR)
  2-2:1.0         (IF) 08:06:50 2EPs () usb-storage host9 (usb1)

robertklep, 2023-05-16:
If there's no output, the correct modules aren't loaded. The provided script should load them, but you can try manually:
```
sudo insmod /lib/modules/usbserial.ko
sudo insmod /lib/modules/pl2303.ko
```

mgov1, 2023-05-16:
$ sudo insmod /lib/modules/usbserial.ko 
Password: 
insmod: ERROR: could not insert module /lib/modules/usbserial.ko: File exists
$ sudo insmod /lib/modules/pl2303.ko 
insmod: ERROR: could not insert module /lib/modules/pl2303.ko: File exist

robertklep, 2023-05-16:
Okay, so that means the drivers have loaded properly.

If that doesn't yield any entries in `/dev`, it means your device isn't supported. I don't know why, but it could be that it requires a more recent driver than Synology provides.

mgov1, 2023-05-16:
Thanks @robertklep, that might explain indeed.
I guess the drivers for DSM 7.2 are the same?

robertklep, 2023-05-16:
Yes, just compiled against a different toolchain version.

mgov1, 2023-05-16:
On a windows system I just found that it's a (Prolific) PL2303GT device.
I wonder if more people have had problems with this.

robertklep, 2023-05-17:
@mgov1 I found [this post](https://patchwork.kernel.org/project/linux-usb/patch/20190213123000.4656-1-charlesyeh522@gmail.com/) which suggests that the PL2023GT does indeed require an updated driver.

I'll try and see if I can manually update the driver.

robertklep, 2023-05-17:
@mgov1 I patched the PL2303 driver to add support for your device. If you want to test it, you can download it from here: https://github.com/robertklep/dsm7-usb-serial-drivers/raw/main/modules/r1000/dsm-7.1/pl2303-hxn.ko

Perhaps you need to rename it back to `pl2303.ko`, not sure.

Big fat disclaimer: I'm not able to test this driver, so I can't guarantee it'll work or not crash your system. I would suggest not copying it to `/lib/modules` until you've tested it out first.

mgov1, 2023-05-17:
Thanks @robertklep 
I’ll verify this evening and let you know.

mgov1, 2023-05-17:
Hi @robertklep,
I can confirm the driver is working.
From dmesg:

_[80723.846360] usb 1-3: new full-speed USB device number 4 using xhci_hcd
[80724.017909] pl2303 1-3:1.0: pl2303 converter detected
[80724.023914] usb 1-3: pl2303 converter now attached to ttyUSB0_

_$ ls -l /dev | grep -i usb
crw------- 1 root root  188,   0 May 17 19:23 ttyUSB0_

Many thanks for fixing this!

robertklep, 2023-05-17:
Cool! Thanks for testing 👍🏻

closed 2023-05-17 by robertklep