robertklep/dsm7-usb-serial-drivers #88: DS 923+ G1000 rndis_host.ko: Unknown symbol in module

hulst88, 2023-05-18:
I am getting the following error and followed all instructions:
rndis_host.ko: Unknown symbol in module

Any help would be awesom to fix this file

robertklep, 2023-05-18:
I think the RNDIS driver depends on other modules that aren't available (anymore) on Synology.

What do you intend to use this driver for?

hulst88, 2023-05-18:
I want to use Sonoff zigbee adaptor for Home Assistant. But it is unable to
recognize the usb and why mqtt docker is not starting

robertklep, 2023-05-18:
You don't need the RNDIS driver for that device, it's either the (built-in) `cdc-acm.ko` driver or the `ch341.ko` driver for your NAS (which is R1000, not G1000).

hulst88, 2023-05-18:
Both are running. Maybe I only need to use 1 of them?

robertklep, 2023-05-18:
It doesn't matter if you loaded them both.

What does the following show?
```
/usr/syno/bin/lsusb -cui
```

hulst88, 2023-05-18:
|__usb1          1d6b:0002:0404 09  2.00  480MBit/s 0mA 1IF  (Linux 4.4.180+ xhci-hcd xHCI Host Controller 0000:06:00.3) hub
  |__1-1         f400:f400:0100 00  2.00  480MBit/s 200mA 1IF  (Synology DiskStation 6500290BA92DF911)
  1-1:1.0         (IF) 08:06:50 2EPs () usb-storage host7 (synoboot)
  |__1-3         1a86:55d4:0442 02  1.10   12MBit/s 132mA 2IFs (ITEAD SONOFF Zigbee 3.0 USB Dongle Plus V2 20230218161535)
  1-3:1.0         (IF) 02:02:01 1EP  () cdc_acm tty/ttyACM0
  1-3:1.1         (IF) 0a:00:00 2EPs () cdc_acm

robertklep, 2023-05-18:
That means the device is available as `/dev/ttyACM0`

hulst88, 2023-05-18:
Thnx Robert, not knowing what I do wrong but I need to learn more I guess.
I will keep researching.

hulst88, 2023-05-19:
After rebooting NAS (update) the device is not available at /dev/ttyACM0 anymore

masterkevin@Opslaghok:~$ /usr/syno/bin/lsusb -cui
|__usb1          1d6b:0002:0404 09  2.00  480MBit/s 0mA 1IF  (Linux 4.4.180+ xhc                                            i-hcd xHCI Host Controller 0000:06:00.3) hub
  |__1-1         f400:f400:0100 00  2.00  480MBit/s 200mA 1IF  (Synology DiskSta                                            tion 6500290BA92DF911)
  1-1:1.0         (IF) 08:06:50 2EPs () usb-storage host7 (synoboot)
  |__1-3         1a86:55d4:0442 02  1.10   12MBit/s 132mA 2IFs (ITEAD SONOFF Zig                                            bee 3.0 USB Dongle Plus V2 20230218161535)
  1-3:1.0         (IF) 02:02:01 1EP  ()
  1-3:1.1         (IF) 0a:00:00 2EPs ()

robertklep, 2023-05-19:
That means the module hasn't been loaded at boot. Did you follow [all installation instructions](https://github.com/robertklep/dsm7-usb-serial-drivers#installation)?

closed 2025-11-18 by robertklep