robertklep/dsm7-usb-serial-drivers #89: DSM 7.1 DS2125J

michaeldn, 2023-05-23:
I have a DS215J with the Armada375. I followed your instructions, but my device (RFLink - Adrino Mega) is on the USB, but not read via the serial port.

I am not certain what is not correct.

robertklep, 2023-05-23:
Can you run this command and post its output here:
```
/usr/syno/bin/lsusb -cui
```

michaeldn, 2023-05-23:
usbserial              24729  1 ftdi_sio
usbcore               151242  10 ehci_hcd,xhci_hcd,etxhci_hcd,usb_storage,usblp,usbhid,usbserial,ftdi_sio,cdc_acm

robertklep, 2023-05-23:
`ch341` should be in that list if it was loaded properly.

michaeldn, 2023-05-23:
Same result - should I reboot the NAS?

robertklep, 2023-05-23:
What happens when you load `ch341.ko` manually?
```
sudo insmod /lib/modules/ch341.ko
```

michaeldn, 2023-05-23:
`insmod: ERROR: could not insert module /lib/modules/ch341.ko: Invalid module format`

robertklep, 2023-05-23:
That means you didn't follow [the download directions](https://github.com/robertklep/dsm7-usb-serial-drivers#downloading-a-module) correctly.

The correct download link is this one: https://github.com/robertklep/dsm7-usb-serial-drivers/raw/main/modules/armada375/dsm-7.0/ch341.ko

michaeldn, 2023-05-23:
That worked thank you! One more question: how do I find the port?

`dmesg | grep tty
[    0.000000] Kernel command line: console=ttyS0,115200 ip=off initrd=0x8000000 root=/dev/md0 rw syno_hw_version=DS215jv10 hd_power_on_seq=2 ihd_num=2 netif_num=1 flash_size=8
[    2.859992] serial8250.0: ttyS0 at MMIO 0xf1012000 (irq = 44) is a 16550A
[    3.586761] console [ttyS0] enabled
[    3.619724] serial8250.1: ttyS1 at MMIO 0xf1012100 (irq = 45) is a 16550A
[   44.658706] synobios open /dev/ttyS1 success
[   61.310107] systemd[1]: Created slice system-serial\x2dgetty.slice.
[   61.317162] systemd[1]: Starting system-serial\x2dgetty.slice.
[   63.330304] synobios open /dev/ttyS1 success
[515201.623975] usb 2-1: ch341-uart converter now attached to ttyUSB0`

robertklep, 2023-05-23:
It should be available as `/dev/ttyUSB0`

michaeldn, 2023-05-27:
Hi - after enabling logging and checking everything I realize I am receiving: `Permission denied: '/dev/ttyUSB0' ` 

Do you have any advice on resolving this? Thank you again.

robertklep, 2023-05-28:
@michaeldn

```
sudo chmod 666 /dev/ttyUSB0
```

michaeldn, 2023-05-30:
Thank you! My Unix memory is from over a decade ago.

closed 2023-05-30 by michaeldn