robertklep/dsm7-usb-serial-drivers #92: Google Coral TPU Driver

nmajin, 2023-06-21:
Is it possible to get this for `synology_avoton` kernel `3.10.108` please? :)

https://coral.ai/docs/m2/get-started/#2a-on-linux

nmajin, 2023-06-21:
This would be for the PCIe TPU. The Synology actually finds the TPU with a Dual Edge PCIe TPU Adapter:
```
$ lspci -nn | grep 089a
03:00.0 Class [0000]: Device [1ac1:089a]
04:00.0 Class [0000]: Device [1ac1:089a]
```
Just needs the driver:
```
$ ls /dev/apex_0
ls: cannot access '/dev/apex_0': No such file or directory
```

robertklep, 2023-06-21:
Sorry, this repository is for simple serial drivers only 🥴

nmajin, 2023-06-21:
Darn, I tried :P

Would you happen to know if this is even possible? I would like to give this a shot by myself, but I just don't know if there is a technical limitation.

robertklep, 2023-06-21:
I think it's probably possible, just outside of my range of expertise 😬

If your intention is to use it with Frigate, have you seen this? https://github.com/weltenwort/frigate-synology-dsm7

Alternatively, if your platform supports it, you can try getting it to work in a virtual machine, which wouldn't require host (that is, DSM) support for a device.

nmajin, 2023-06-21:
Thanks for sharing, but that is for the USB Coral, I have the M.2 TPU...I've hit the max USB power and ports on my Synology haha, really trying to use every bit of the thing I can :P

I do know about VMM as well...but I am all in with my docker instances, but I may play around with that more. Thanks again for the info.

robertklep, 2023-06-21:
Ohh right I wondered what the PCIe part was all about, M.2 makes sense now ;)

I have no idea if DSM even has generic M.2 support, or if it's possible to pass through to a VMM somehow.