robertklep/dsm7-usb-serial-drivers #95: Failure to compile for 7.2 Broadwell

mircevski, 2023-06-27:
Hello,

I've followed the tutorial on how to compile, I've setup my build environment on an Ubuntu "22.04.1 LTS (Jammy Jellyfish)", as follows:

> mkdir -p /toolkit
> cd /toolkit/
> git clone https://github.com/SynologyOpenSource/pkgscripts-ng
> apt install cifs-utils python python-pip python3 python3-pip
> apt update
> apt install cifs-utils python python-pip python3 python3-pip
> apt install cifs-utils  python3 python3-pip
> cd pkgscripts-ng/
> git checkout DSM7.2
> ./EnvDeploy -v 7.2 -p broadwell

I've downloaded the usb serial drivers 
> git clone https://github.com/robertklep/dsm7-usb-serial-drivers and only kept the _4.4.x_ folder into _/toolkit/source/_

As I can see, the first few make lines are going fine, then one tmp file is reported as missing (.ch341.o.tmp) then everything breaks. My programming knowledge is minimal. Can anyone help ?

I'm attaching the log.
[compile.log](https://github.com/robertklep/dsm7-usb-serial-drivers/files/11880847/compile.log)

robertklep, 2023-06-27:
Compiling these modules is a PITA, I also get the error you're getting a lot. Sometimes just trying to compile the modules again works.

Are you trying to compile a module that's not provided? If so, I'd be happy to take a look at it.

mircevski, 2023-06-27:
Actually, by removing parallelization, by inserting "MAKEFLAGS := -j 1" in Makefile, things compile perfectly. So, I just need to figure out the if the .SPK package actually contains all drivers and if they actually load into the kernel.

mircevski, 2023-06-27:
The spk seems a generic build, I think it is not usable as it is.
The real problem is now that the modules do not load, due to kernel mismatch: "insmod: ERROR: could not insert module /lib/modules/pl2303.ko: Unknown symbol in module"

# uname -a
Linux XXXXX 4.4.302+ #64570 SMP Tue Jun 13 21:40:16 CST 2023 x86_64 GNU/Linux synology_broadwell_3617xs

At this point I think that the problem is in the environment setup and somehow I did not get correct platform/kernel headers version. But what I've prepared the environment for is the 7.2 broadwell as per the following:
> git checkout DSM7.2
> ./EnvDeploy -v 7.2 -p broadwell

Is the 7.2 version enough or we need to be more specific versions of the build environment that match exactly the current DSM OS version, like the minor version and/or the build number.

One note, the Ubuntu I've used to build the modules was 22.04, and it actually run under Windows WSL, which for me should not matter, but anyway.

Thanks

mircevski, 2023-06-27:
... after some more research, I could find that the version od the build environment is mismatching the actual DSM version installed, by  the buildnumber. While the wodnloaded packed is as follows ...

> # cat /toolkit/build_env/ds.broadwell-7.2/PkgVersion
> majorversion="7"
> minorversion="2"
> major="7"
> minor="2"
> micro="0"
> buildphase="Dev"
> buildnumber="63133"
> smallfixnumber="0"
> nano="0"
> base="63133"
> productversion="7.2"

... my build number on the DS3617xs is a higher one: 64570 

Maybe Synology didn't release the latest packages ?
Any help will be much appreciated :)

Regards,

robertklep, 2023-06-27:
> The real problem is now that the modules do not load, due to kernel mismatch: "insmod: ERROR: could not insert module /lib/modules/pl2303.ko: Unknown symbol in module"

Are you sure that's because of a kernel mismatch and not because you didn't load the requirements (`usbserial.ko`)?

mircevski, 2023-06-28:
you ar wright, loading the usbserial.ko as prerequisite worked