robertklep/dsm7-usb-serial-drivers #96: DSM 7.2-64570 Update 1 not able to install module with error Invalid Module Format

jeenyeong, 2023-06-28:


robertklep, 2023-06-28:
Please follow the [download instructions](https://github.com/robertklep/dsm7-usb-serial-drivers#downloading-a-module).

You're downloading the HTML page, not the module itself.

closed 2023-06-28 by robertklep

jeenyeong, 2023-06-28:
@robertklep 

Thanks and it works great for now :)

```
jeenyeong@Jeen-NAS:/lib/modules$ sudo rm ch341.ko
jeenyeong@Jeen-NAS:/lib/modules$ sudo rm cp210x.ko
jeenyeong@Jeen-NAS:/lib/modules$ sudo rm pl2303.ko
jeenyeong@Jeen-NAS:/lib/modules$ sudo rm ti_usb_3410_5052.ko
jeenyeong@Jeen-NAS:/lib/modules$ sudo wget https://github.com/robertklep/dsm7-usb-serial-drivers/raw/main/modules/v1000/dsm-7.2/ch341.ko
--2023-06-29 02:46:38--  https://github.com/robertklep/dsm7-usb-serial-drivers/raw/main/modules/v1000/dsm-7.2/ch341.ko
Resolving github.com... 20.205.243.166
Connecting to github.com|20.205.243.166|:443... connected.
HTTP request sent, awaiting response... 302 Found
Location: https://raw.githubusercontent.com/robertklep/dsm7-usb-serial-drivers/main/modules/v1000/dsm-7.2/ch341.ko [following]
--2023-06-29 02:46:39--  https://raw.githubusercontent.com/robertklep/dsm7-usb-serial-drivers/main/modules/v1000/dsm-7.2/ch341.ko
Resolving raw.githubusercontent.com... 185.199.108.133, 185.199.111.133, 185.199.110.133, ...
Connecting to raw.githubusercontent.com|185.199.108.133|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 255928 (250K) [application/octet-stream]
Saving to: 'ch341.ko'

ch341.ko                                                                                                                                                        100%[====================================================================================================================================================================================================================================================================================================================================================================================================================================================>] 249.93K  1.32MB/s    in 0.2s

2023-06-29 02:46:39 (1.32 MB/s) - 'ch341.ko' saved [255928/255928]

jeenyeong@Jeen-NAS:/lib/modules$ sudo insmod /lib/modules/ch341.ko
jeenyeong@Jeen-NAS:/lib/modules$ sudo wget https://github.com/robertklep/dsm7-usb-serial-drivers/raw/main/modules/v1000/dsm-7.2/cp210x.ko
--2023-06-29 02:46:49--  https://github.com/robertklep/dsm7-usb-serial-drivers/raw/main/modules/v1000/dsm-7.2/cp210x.ko
Resolving github.com... 20.205.243.166
Connecting to github.com|20.205.243.166|:443... connected.
HTTP request sent, awaiting response... 302 Found
Location: https://raw.githubusercontent.com/robertklep/dsm7-usb-serial-drivers/main/modules/v1000/dsm-7.2/cp210x.ko [following]
--2023-06-29 02:46:49--  https://raw.githubusercontent.com/robertklep/dsm7-usb-serial-drivers/main/modules/v1000/dsm-7.2/cp210x.ko
Resolving raw.githubusercontent.com... 185.199.109.133, 185.199.110.133, 185.199.111.133, ...
Connecting to raw.githubusercontent.com|185.199.109.133|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 272960 (267K) [application/octet-stream]
Saving to: 'cp210x.ko'

cp210x.ko                                                                                                                                                       100%[====================================================================================================================================================================================================================================================================================================================================================================================================================================================>] 266.56K  --.-KB/s    in 0.1s

2023-06-29 02:46:50 (1.92 MB/s) - 'cp210x.ko' saved [272960/272960]

jeenyeong@Jeen-NAS:/lib/modules$ sudo insmod /lib/modules/cp210x.ko
jeenyeong@Jeen-NAS:/lib/modules$ sudo wget https://github.com/robertklep/dsm7-usb-serial-drivers/raw/main/modules/v1000/dsm-7.2/pl2303.ko
--2023-06-29 02:46:57--  https://github.com/robertklep/dsm7-usb-serial-drivers/raw/main/modules/v1000/dsm-7.2/pl2303.ko
Resolving github.com... 20.205.243.166
Connecting to github.com|20.205.243.166|:443... connected.
HTTP request sent, awaiting response... 302 Found
Location: https://raw.githubusercontent.com/robertklep/dsm7-usb-serial-drivers/main/modules/v1000/dsm-7.2/pl2303.ko [following]
--2023-06-29 02:46:57--  https://raw.githubusercontent.com/robertklep/dsm7-usb-serial-drivers/main/modules/v1000/dsm-7.2/pl2303.ko
Resolving raw.githubusercontent.com... 185.199.108.133, 185.199.109.133, 185.199.110.133, ...
Connecting to raw.githubusercontent.com|185.199.108.133|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 278384 (272K) [application/octet-stream]
Saving to: 'pl2303.ko'

pl2303.ko                                                                                                                                                       100%[====================================================================================================================================================================================================================================================================================================================================================================================================================================================>] 271.86K  1.39MB/s    in 0.2s

2023-06-29 02:46:58 (1.39 MB/s) - 'pl2303.ko' saved [278384/278384]

jeenyeong@Jeen-NAS:/lib/modules$ sudo insmod /lib/modules/pl2303.ko
[... cut ...] ==================================================================================================================================================================================================================================================================================================================>] 303.77K  1.37MB/s    in 0.2s

2023-06-29 02:47:06 (1.37 MB/s) - 'ti_usb_3410_5052.ko' saved [311056/311056]

jeenyeong@Jeen-NAS:/lib/modules$ sudo insmod /lib/modules/ti_usb_3410_5052.ko

```